ros-navigation/navigation2 #714: Scrub parameter declaration in plugins

mjeronimo, 2019-05-14:
## Feature request

#### Feature description
Parameters can only be declared once per node. Normally, they are declared in the node's constructor. However, in the case of plug-in modules which are not themselves nodes, but use the parent node handle, the plugin's parameters are declared against the parent's node. Therefore, if a plugin is reset, uninitialized and reset, it should probably undeclare its parameters and redeclare them upon reinitialization.

bpwilcox, 2019-05-14:
The ClearCostmap recovery behavior should be removed from behavior tree until this issue is resolved, otherwise there will be runtime errors when resetting the layer (at least static layer).

SteveMacenski, 2019-05-14:
Plugins in costmap_2d have been often reset as a recovery. Would reinitializing the variables on reset incur any significant delay in operations since this is a pretty regular occurrence?

orduno, 2019-05-24:
@SteveMacenski It's possible, we would have to test it. Perhaps there's no need to un-declare the parameters upon reset. Could you review #752? That should fix, at least temporarily, the bug.

bpwilcox, 2019-10-18:
Since PR #1099, parameter re-declarations are being handled in most cases by using the added `declare_if_not_declared` utility function. The alternative seemingly suggested in this issue is to undeclare the parameters when the plugins are destroyed, as opposed to leaving them on the node. 

Are there any strong reasons we should move to specifically undeclare all parameters of a plugin (or any object declaring parameters during the `on_configure` call)? If there aren't any strong cases, I think the current solution resolves this ticket. @orduno

SteveMacenski, 2019-10-18:
I'm happy with that. I think undeclaring is a bit excessive but if someone's a completeist I would understand.

Functionally, I dont know when I'm regularly loading and unloading plugins outside of restarting a process

closed 2019-11-11 by SteveMacenski