ros-navigation/navigation2 #728: ERROR ... does not name a type NodeOptions

jajosheni, 2019-05-16:
For Bug report or feature requests, please fill out the relevant category below
-->

## Bug report

**Required Info:**

- Operating System: Ubuntu 18.04 beaver 64bit
- Installation type: debian Packages
- Version or commit hash: https://raw.githubusercontent.com/ROBOTIS-GIT/turtlebot3/ros2/turtlebot3.repos
- DDS implementation: Fast-RTPS
- Client library (if applicable): rclcpp

#### Steps to reproduce issue

```
colcon build --symlink-install
```

#### Expected behavior
Successful build
#### Actual behavior

--- stderr: nav2_util                                                                                               
In file included from /home/lab/turtlebot3_ws/src/navigation2/navigation2/nav2_util/include/nav2_util/service_client.hpp:20:0,
                 from /home/lab/turtlebot3_ws/src/navigation2/navigation2/nav2_util/include/nav2_util/lifecycle_service_client.hpp:22,
                 from /home/lab/turtlebot3_ws/src/navigation2/navigation2/nav2_util/src/lifecycle_utils.cpp:20:
/home/lab/turtlebot3_ws/src/navigation2/navigation2/nav2_util/include/nav2_util/node_utils.hpp:73:9: error: ‘NodeOptions’ in namespace ‘rclcpp’ does not name a type
 rclcpp::NodeOptions
         ^~~~~~~~~~~
make[2]: *** [CMakeFiles/nav2_util_core.dir/src/lifecycle_utils.cpp.o] Error 1
make[1]: *** [CMakeFiles/nav2_util_core.dir/all] Error 2
make: *** [all] Error 2
---

#### Additional information
I'm using Crystal and i Want to install Turtlebot3 packages but i cannot get a clean build from colcon. Please Help!

crdelsey, 2019-05-16:
The turtlebot instructions pull the master branch of the navigation2 stack. It is compatible with Dashing, not Crystal.

To get the Crystal version, you'll want to change to the `/home/lab/turtlebot3_ws/src/navigation2/navigation2` directory and `git checkout crystal-devel`

LeonJung, 2019-05-17:
refer #682

corsaircpt, 2019-06-13:
Looks like navigation2 is also not compatible with dashing either. I have dashing compiled and installed and pulled in master of navigation2 with the Turtlebot3 package but hit the below issue

Finished <<< nav2_voxel_grid [1min 14s]                                        
Starting >>> dwb_msgs
[Processing: dwb_msgs, nav2_msgs]                                              
[Processing: dwb_msgs, nav2_msgs]                                              
[Processing: dwb_msgs, nav2_msgs]                                              
[Processing: dwb_msgs, nav2_msgs]                                              
[Processing: dwb_msgs, nav2_msgs]                                              
[Processing: dwb_msgs, nav2_msgs]                                              
[Processing: dwb_msgs, nav2_msgs]                                              
[Processing: dwb_msgs, nav2_msgs]                                              
[Processing: dwb_msgs, nav2_msgs]                                              
[Processing: dwb_msgs, nav2_msgs]                                              
Finished <<< dwb_msgs [5min 6s]                                                
Starting >>> angles
Finished <<< angles [10.0s]                                                    
Starting >>> behaviortree_cpp
[Processing: behaviortree_cpp, nav2_msgs]                                      
[Processing: behaviortree_cpp, nav2_msgs]                                      
[Processing: behaviortree_cpp, nav2_msgs]    
[Processing: behaviortree_cpp, nav2_msgs]    
[Processing: behaviortree_cpp, nav2_msgs]                                      
Finished <<< nav2_msgs [11min 49s]           
Starting >>> nav2_util
Finished <<< behaviortree_cpp [2min 43s]    
Starting >>> nav_2d_utils
--- stderr: nav2_util                                                  
CMake Error at CMakeLists.txt:26 (install):
  install TARGETS given target "nav2_util_core" which does not exist in this
  directory.

---
Failed   <<< nav2_util	[ Exited with code 1 ]
Aborted  <<< nav_2d_utils

crdelsey, 2019-07-08:
@corsaircpt @jajosheni I'm sorry things have been very unstable recently. ROS 2, Navigation 2 and Turtlebot 3 have all been struggling to stay in sync as an API change makes it's way through the stack.

I believe we can fix that in the future by having users track the latest release branch instead of master. I believe you can use the dashing branch of navigation 2 and ros 2 with the current turtlebot 3 code and everything will now build fine.

Are you still having any problems? We should be able to solve them now.

corsaircpt, 2019-07-11:
ok will try this soon from scratch with a fresh pull and update.

crdelsey, 2019-07-29:
I believe build issues are fixed and stable if users stick to the ROS 2 and nav2 dashing releases. Closing this issue.

closed 2019-07-29 by crdelsey