ros-navigation/navigation2 #729: Error in ros2 launch gazebo_rviz2_launch.py

lbegani, 2019-05-16:
- Operating System: Ubuntu 18.10
- Installation type: source (both navigation2 and ROS2)
- Version or commit hash: de07f90aa8f67cc5e508d6928f3659d54dce9cf5 (navigation2 latest commit on master, ROS2 latest commit on master)

When I try to launch gazebo_rviz2_launch.py, I get the following errors - 

> lkumarbe@intel:~$ ros2 launch nav2_bringup gazebo_rviz2_launch.py world:=/home/lkumarbe/turtlebot3_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_worlds/burger.model
[INFO] [launch]: All log files can be found below /home/lkumarbe/.ros/log/2019-05-15-15-28-09-982646-intel-17338
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gazebo-1]: process started with pid [17355]
[INFO] [rviz2-2]: process started with pid [17356]
[gazebo-1] Gazebo multi-robot simulator, version 9.6.0
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[gazebo-1] Gazebo multi-robot simulator, version 9.6.0
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[rviz2-2] [INFO] [rviz2]: Stereo is NOT SUPPORTED
[rviz2-2] [INFO] [rviz2]: OpenGl version: 3 (GLSL 1.3)
[rviz2-2] [WARN] [rviz2]: Could not find folder /home/lkumarbe/ros2_ws/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/scripts
[rviz2-2] [WARN] [rviz2]: Could not find folder /home/lkumarbe/ros2_ws/install/rviz_rendering_tests/share/rviz_rendering_tests/ogre_media_resources/meshes
[rviz2-2] [INFO] [rviz2]: Stereo is NOT SUPPORTED
**[gazebo-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidArgument'
[gazebo-1]   what():  failed to initialize rcl node: name argument is null, at /home/lkumarbe/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:136
[ERROR] [gazebo-1]: process has died [pid 17355, exit code 255, cmd 'gazebo --verbose -s libgazebo_ros_init.so** /home/lkumarbe/turtlebot3_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_worlds/burger.model'].

mkhansen-intel, 2019-05-24:
Did you set the GAZEBO_MODEL_PATH?
```
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:<full/path/to/my_robot/models>
```

mkhansen-intel, 2019-05-31:
I didn't hear back on this, is it resolved? If so please update this issue and close it.

lbegani, 2019-06-05:
Sorry for delayed response. I was waiting for the Dashing release to happen and test it again. I found few things broken in between. I will close it if the issue is not seen in the latest release.

lbegani, 2019-06-06:
I see the same error in the latest Dashing ROS2 (release-latest) + latest Navigation2 (51bbc3bf9fc084491e66ad46a23bca1ea3cdc123). I will try to rootcause the issue. Meanwhile, below is the latest error log - 

```
lkumarbe@intel:~$ source ~/ros2_ws/install/local_setup.bash
lkumarbe@intel:~$ source ~/navigation2_ws/install/local_setup.bash
lkumarbe@intel:~$ source ~/navstack_dependencies_ws/install/local_setup.bash
lkumarbe@intel:~$ source ~/turtlebot3_ws/install/local_setup.bash
lkumarbe@intel:~$ export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/lkumarbe/turtlebot3_ws/src/turtlebot3_simulations/turtlebot3_gazebo/models/
lkumarbe@intel:~$ ros2 launch nav2_bringup gazebo_rviz2_launch.py world:=/home/lkumarbe/turtlebot3_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_worlds/burger.model
[INFO] [launch]: All log files can be found below /home/lkumarbe/.ros/log/2019-06-06-10-43-21-657063-intel-24762
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [gazebo-1]: process started with pid [24779]
[INFO] [rviz2-2]: process started with pid [24780]
[gazebo-1] Gazebo multi-robot simulator, version 9.6.0
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[gazebo-1] Gazebo multi-robot simulator, version 9.6.0
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[gazebo-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLInvalidArgument'
[gazebo-1]   what():  failed to initialize rcl node: name argument is null, at /home/lkumarbe/ros2_ws/src/ros2/rcl/rcl/src/rcl/node.c:136
[ERROR] [gazebo-1]: process has died [pid 24779, exit code 255, cmd 'gazebo --verbose -s libgazebo_ros_init.so /home/lkumarbe/turtlebot3_ws/src/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_worlds/burger.model'].
```

lbegani, 2019-06-06:
I have updated turtlebot3 pacakges and this seems to have fixed the issue.

closed 2019-06-06 by lbegani