ros-navigation/navigation2 #730: Work required on system and module-level tests

mjeronimo, 2019-05-16:
There have been several changes recently: ROS2 actions, lifecycle nodes, ROS2 parameter API changes, etc. There is some work required in the tests to accomodate these changes. All of these changes are being integrated on the 'lifecycle' branch first and wil then be integrated into the master branch. The known issures are as follows:

- [ ] **nav2_system_tests/src/localization** - This test currently builds and runs, but AMCL is outputting an initial pose of 0,0,0, which the test does not expect. The lifecycle node implementation of AMCL saves its last known pose, even when the node is brought back to the unconfigured state. This way, when it starts again, it remembers where it was. However, when AMCL first comes up, should it immediately output its "last known state" of 0,0,0 or wait until it moves again? 
	
- [ ] **nav2_system_tests/src/planner** - This test needs to be ported to ROS2 actions. It also needs the lifecycleManager to be added to the launch script. 
	
- [ ] **nav2_system_tests/src/system** - This test is outputting the initial pose before the AMCL node is fully activated, resulting in AMCL missing the initial pose. Simple navigator should be removed from any launch scripts since this node has been deleted. 
	
- [ ] **nav2_system_tests/src/dummy_controller** - This needs to be ported to ROS2 Actions. 
	
- [ ] **nav2_system_tests/src/dummy_planner** - This needs to be ported to ROS2 Actions. 
	
- [ ] **nav2_system_tests/src/updown** - This is a stress test that doesn't need to be gtest-enabled. It won't be run as part of CI, but could be used in a batch of stress tests. 
	
- [ ] **nav2_dwb_controller/dwb_plugins/test/goal_checker** - The StoppedGoalChecker derives from the SimpleGoalChecker. However, both are used in the test app, resulting in a declare_parameters error (each parameter can only be declared once). We need to resolve how parameters are handled in plug-ins. 
	
- [ ] **nav2_dwb_controller/dwb_plugins/test/twist_gen** - The KinematicParameters is used in multiple places, resulting in redeclared parameter errors. Similar to the previous issue, we need to resolve how parameters are handled in plug-ins and related classes lik this. 
	
- [ ] **nav2_costmap_2d/test/integration/obstacle_tests** - This test needs to have the lifecycle manager added to automatically handle the lifecycle nodes. 
	
- [ ] **nav2_costmap_2d/test/integration - These tests (footprint_tests, static_tests, and inflation tests) are not enabled. They are not enabled on the master branch as well.

crdelsey, 2019-07-08:
- Remove dummy planner, controller since they aren't currently used.
- Make sure there is an issue for goal checker test.
- Then close this issue.

crdelsey, 2019-10-02:
Obsolete

closed 2019-10-02 by crdelsey