ros-navigation/navigation2 #735: AMCL fails to localize in simulation environment

cwbollinger, 2019-05-17:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source (ros2 and navigation2)
- Version or commit hash:
  - navigation2 master branch 178c7c6ce4aa3ea58189302e1287d44f800543cd
  - ros2 in sync with master (wget ros2.repos / vcs import src) as of ~5:00 PM 6/16/2018
- DDS implementation:
  - fastrtps
- Client library (if applicable):
  - NA

#### Steps to reproduce issue
- Follow steps according to https://github.com/ros-planning/navigation2/blob/master/nav2_bringup/README.md
    - Launch nav2_bringup/gazebo_rviz2_launch.py
    - Launch turtlebot3_bringup/turtlebot_state_publisher.launch.py w/ waffle configuration
    - Launch nav2_bringup/nav2_bringup_1st_launch.py

#### Expected behavior
AMCL receives laser scan and tf, and localizes the robot when provided an initial pose by rviz. (Note, I was able to get this behavior about a week ago. Unfortunately I don't know the exact commit)

#### Actual behavior
- After launching gazebo and state_publisher
    - Can visualize laser scans in rviz, scans are updating as expected.
    -  odom -> base_link -> laser_scan tf tree is being published. Can visualize the tf tree in rviz. (map -> odom doesn't exist yet as amcl isn't running). So far so good

- After launching nav2_bringup_1st_launch.py
    - Map appears in rviz
    - amcl log: ```[INFO] [amcl]: Received a 128 X 122 map @ 0.050 m/pix```
    - using rviz localization tool: ```[INFO] [amcl]: Setting pose (1558026564.674396): 0.062 -0.277 0.051```. However, the map -> odom transform is still not being published (?)
    - After this, repeatedly  ```[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 0.201 for reason(0) ``` every 0.2 seconds until ^C

- This seems like a timing issue even though use_sim_time:=True is set as a launchfile argument, so I attempted to run ```ros2 param set /amcl use_sim_time True```. This times out without getting the normal "success" message.
    - ros2 daemon stop/start does not fix this

I've tried the launch procedure on two different lab computers here, and get the same results.

I'm not certain this is actually an issue with nav2 as opposed to my simulation config, but I'm a bit stumped about where things are going wrong. Any guidance would be greatly appreciated!

#### Additional information
Full AMCL log:
```
[ERROR] [amcl]: Invalid type 'string' for parameter 'use_sim_time', should be 'bool'
[INFO] [amcl]: Initializing AMCL
[INFO] [amcl]: Sensor model type is: "likelihood_field"
[INFO] [amcl]: Robot motion model is differential
[WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[ERROR] [amcl]: Invalid type 'string' for parameter 'use_sim_time', should be 'bool'
[INFO] [amcl]: Subscribed to map topic.
[INFO] [amcl]: Subscribing to topic: /parameter_events
[WARN] [amcl]: Parameter '/amcl/resample_interval' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/transform_tolerance' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/laser_max_beams' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/odom_alpha1' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/odom_alpha2' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/odom_alpha3' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/odom_alpha4' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/odom_alpha5' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/laser_z_hit' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/laser_z_short' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/laser_z_max' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/laser_z_rand' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/laser_sigma_hit' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/laser_lambda_short' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/kld_err' is either unregistered or not set
[WARN] [amcl]: Parameter '/amcl/kld_z' is either unregistered or not set
[... 34 lines cut ...]
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 5.400 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 5.600 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 5.800 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 6.000 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 6.200 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 6.400 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 6.600 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 6.801 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 7.001 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 7.200 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 7.401 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 7.601 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 7.800 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 8.000 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 8.201 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 8.400 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 8.600 for reason(0)
[WARN] [amcl]: Laser scan has not been received (and thus no pose updates have been published). Verify that data is being published on the scan topic.
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 8.800 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 9.000 for reason(0)
... (same message repeats for some time) ...
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 22.600 for reason(0)
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 22.800 for reason(0)
[INFO] [rclcpp]: signal_handler(signal_value=2)
```

bergercookie, 2019-05-17:
Probably this is related to https://github.com/ros-planning/navigation2/issues/707

mhpanah, 2019-05-17:
@cwbollinger Try launching the rest of stack and see if that fixes the issue.  I think if you set the use_sim_time_true to true manually that should resolve the issue. 
1. Launch nav2_bringup/gazebo_rviz2_launch.py
2. Launch turtlebot3_bringup/turtlebot_state_publisher.launch.py w/ waffle configuration
3. Launch nav2_bringup/nav2_bringup_1st_launch.py
4. ros2 param set /world_model use_sim_time true; ros2 param set /global_costmap/global_costmap use_sim_time true; ros2 param set /local_costmap/local_costmap use_sim_time true
5. rviz
6. set initial pose from rviz

cwbollinger, 2019-05-18:
@mhpanah Thanks for taking a look. I did some more testing today, and it's definitely a timestamp issue. I think your directions were a partial success, but haven't fixed everything. Here's what I'm seeing:

1. Launch nav2_bringup/gazebo_rviz2_launch.py
2. Launch turtlebot3_bringup/turtlebot_state_publisher.launch.py w/ waffle configuration
3. Launch nav2_bringup/nav2_bringup_1st_launch.py
    - same issues as before
4. ```ros2 param set /world_model use_sim_time true; ros2 param set /global_costmap/global_costmap use_sim_time true; ros2 param set /local_costmap/local_costmap use_sim_time true```
    - None of these namespaces/nodes exist (I think these nodes are brought up in the 2nd_launch file?). Command reports as such and doesn't (afaik) change anything.
    - Try ```ros2 param set /amcl use_sim_time``` again. Today this did not time out, and when the parameter is set AMCL can localize as expected (hooray!). Seems like that something has changed, not sure why that wasn't working yesterday?

Moving on with the rest of the stack...

5. Launch nav2_bringup/nav2_bringup_2nd_launch.py
    - Lots of timestamp errors as expected, haven't set use_sim_time yet
    - Now I should be able to run your step 4 because these nodes are running
6. ```ros2 param set /world_model use_sim_time true; ros2 param set /global_costmap/global_costmap use_sim_time true; ros2 param set /local_costmap/local_costmap use_sim_time true```
    - Command either hangs or prints ```Wait for serviced timed out``` 3 times. In cases where it hangs I left it running for about 5 minutes before ^C, maybe it requires more time (seems unlikely)?
    - ```ros2 daemon stop/start``` didn't change this behavior
    - I restarted bringup_2nd_launch.py several times, got the same behavior

Based on this, I think my original understanding of my problem was incorrect. The actual issue I'm having should be described as follows:

### New Problem Description
- nav2 launch files are unable to reliably start nodes with use_sim_time set to true.
    - The bug referenced as the reason for this in the bringup guide (https://github.com/ros2/rclcpp/issues/595) was marked fixed as of December, so I'm not clear on why this still necessary.
    - It shouldn't really matter though, since you can fix any unset parameters with ```ros2 param set```
- However, sometimes (or currently for me all the time) ```ros2 param set``` fails to change the parameter values of certain nodes in the navigation stack
  - This manifests as either a service timeout or hangs indefinitely
  - Originally noticed with /acml node, now  also with /global_costmap, /local_costmap, and /world_model

Is this a common issue with ros2 launchfiles/ros2 param calls? If I can't reliably set the use_sim_time parameter, I'm not sure how I should proceed past this. Maybe I can modify the default value of ```use_sim_time``` in the nodes' source as a stopgap measure?

mhpanah, 2019-05-21:
@cwbollinger After running the 2nd launch file and running `ros2 param set /world_model use_sim_time true; ros2 param set /global_costmap/global_costmap use_sim_time true; ros2 param set /local_costmap/local_costmap use_sim_time true` the hanging is expected. After this step you would have to launch rviz and then set the initial pose from rviz.   Once you set the initial pose, the set sim time commend will work.   That's because AMCL is not updating transforms before getting the initial pose.  Once AMCL received it's initial pose then global costmap gets activated.   Please try again in the same order as my previous comment.  I updated the comment so the 2nd launch file is there.  Thanks.

mkhansen-intel, 2019-05-21:
I'm seeing a problem with the latest rmw_fastrtps release (0.7.2) which is causing our system test to hang, you might be seeing the same issue. https://github.com/ros2/rmw_fastrtps/issues/280 The quickest way to check is to try using the OpenSplice RMW implementation. 

See directions for how to do that here:
https://index.ros.org//doc/ros2/Tutorials/Working-with-multiple-RMW-implementations/

You first have to install OpenSplice:
`sudo apt install libopensplice69`

Then follow the directions to rebuild your ros2_ws with --cmake-force-configure

Then 
`export RMW_IMPLEMENTATION=rmw_opensplice_cpp`

Let us know if that resolves your issue. If so, the problem is the same as I linked above.

MengNan-Li, 2019-05-27:
I have the same question in the real robot environment.
[INFO] [amcl]: [signalFailure] Drop message: frame 'base_scan' at time 7.200 for reason(0)

cwbollinger, 2019-05-29:
That seems to have done the trick, thanks! I've been able to run the stack 5 or 6 times without errors using OpenSplice and the instructions as written in the nav2_bringup README. I'll go ahead and close this.

closed 2019-05-29 by cwbollinger