ros-navigation/navigation2 #739: colcon build error

routiful, 2019-05-21:
## Bug report

Failed to build navigation2

- Operating System:
  - Ubuntu 18.04, macos high sierra
- Installation type:
  - source
- Version or commit hash:
  - master(https://github.com/ros-planning/navigation2/commit/178c7c6ce4aa3ea58189302e1287d44f800543cd)
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue
I attached references.
https://github.com/ROBOTIS-GIT/turtlebot3/issues/401#issuecomment-494182638
https://github.com/ROBOTIS-GIT/turtlebot3/issues/424#issue-443821707

#### Expected behavior

#### Actual behavior

#### Additional information

crdelsey, 2019-05-29:
@routiful Sorry for the delay. I was on vacation last week.

There are a lot of different problems listed in those bug reports. None of the problems look like a failure to build the behavior tree library; they are all missing dependencies or incorrect use of the `install/setup.bash` files.

If there is a specific problem, let me know and I can try to fix it. However, to me the real problem is that it is too complicated to build everything from source for the average user.

I think the best solution is for us to keep the crystal-devel branch updated (at least until the Dashing release), and you can just ask your users to `apt install ros-crystal-navigation2` or, if you want to build from source to get the latest features, they can just run `rosdep install ...` before building.

mkhansen-intel, 2019-05-31:
I think this PR https://github.com/ROBOTIS-GIT/turtlebot3/pull/432 submitted to the turtlebot3 repo points users to the crystal-devel branch of navigation2 and other packages. I haven't tested it myself but I think it would be good to do that and test it. That may resolve many user build issues. Once we have all the dashing updates done, we'll create a dashing-devel and you can update to using that or preferrably just install from binaries.

crdelsey, 2019-07-29:
I believe build issues are fixed and stable if the user uses the ROS 2 dashing release and the navigation 2 dashing release. Closing this issue.

closed 2019-07-29 by crdelsey