ros-navigation/navigation2 #742: Replace chrono calls (where applicable) with rclcpp time

SteveMacenski, 2019-05-21:
There is now a pretty decent Time library, duration, and rate objects in ROS2. We should move over to this and abstract out the time client so that it better works in simulation with scaled times.

crdelsey, 2019-06-10:
Being able to use std::chrono_literals for constants is a nice feature. I'd not want to give that up. Probably we'd want a std::chrono version of the function and a rclcpp::Duration version so you can use whichever is more appropriate.

SteveMacenski, 2019-10-15:
I did an audit of these, it seems nearly all the chrono calls are timeouts for IPC calls, throttling logging, or throttling rate controller in BT. I'm going to close this ticket since none of these represent big enough problems if simlation time was scaled (which will happen in simulation). they're not loop rates for controllers or update triggers for costmaps / planners. Though my feeling is that none of the behavior tree stuff respects simulation times correctly, I dont think that represents a safety risk. Just if you're running at 10x or 0.1x speed the behavior is going to be odd from an orchestration standpoint

closed 2019-10-15 by SteveMacenski