ros-navigation/navigation2 #746: Problem of using namespace

LeonJung, 2019-05-22:
## Issue report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - src
- Version or commit hash:
  - modified 0.1.7
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Feature description

As I put namespace on nav2 launches, I found a minor problem and a major problem.

Minor problem is that, the topic "/cmd_vel", for example, has slash in front of the keyword, so the topic couldn't be covered with the namespace. There are also few topics which is the same, but are not in the nav2 stack but are in the ros2 core sources: /tf for instance. Of course you can just get rid of the slash in both cases. 

Major problem is that, currently the ros2 core (?) or what can I say... ros2 launch? doesn't seem support multi-layered namespace, so when local and global costmap nodes are brought on, since they uses the *_costmap as the node name and node namespace, the namespace in ros2 launch overwrites the *_costmap as namespace, and now "namespace_set_in_launch/*_costmap" is happened. As parameter server also is in the "*_costmap" namespace, it disappears as it is covered with "namespace_set_in_launch". 

This issue is for letting know about what will be happened after your development (I hope this is not a talk after the event XD) and **also for asking any ideas for putting namespace safely on nav2 stack**. I'm currently doing this for making multi-robot control, so if anybody has any ideas on that, any suggestion will be a great help.

Cheers

orduno, 2019-07-25:
@LeonJung I'm working on a solution for your problem and will have an update soon.

orduno, 2019-08-23:
@LeonJung Can you please take a look at #1057. It proposes solutions to the problems you mention above.

SteveMacenski, 2019-09-21:
Is this still an open from the pending PRs?

orduno, 2019-10-15:
Closing ticket as multi-robot scenario is now supported.

closed 2019-10-15 by orduno