ros-navigation/navigation2 #753: lifecycle: nav2_bringup relies on turtlebot3 packages

mkhansen-intel, 2019-05-24:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

Running nav2_bringup on the lifecycle branch always requires the turtlebot3_gazebo package, even if not using simulation.
```
ros2 launch nav2_bringup nav2_bringup_launch.py use_simulation:=False
```
#### Expected behavior
User shouldn't need turtlebot3 packages to run our nav2_bringup_launch.

#### Actual behavior
```
[ERROR] [launch]: Caught exception in launch (see debug for traceback): "package 'turtlebot3_gazebo' not found,
````
#### Additional information
We shouldn't launch simulation in the same launch file as the robot stack anyway, because we may be running on a real robot in a headless mode. Simulation and rviz should be launched separately. We have a separate launch file for that.

mkhansen-intel, 2019-05-24:
Also, while looking at this I noticed there's a hard-coded path in the nav2_bringup_launch.py file that needs to be fixed.

crdelsey, 2019-06-27:
Fixed in #841

closed 2019-06-27 by crdelsey