ros-navigation/navigation2 #754: Release build fails for lifecycle branch

mkhansen-intel, 2019-05-24:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - lifecycle branch
- DDS implementation:
  - FastRTPS (doesn't matter)
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue

```
git checkout lifecycle
colcon build --symlink-install
```

#### Expected behavior
The default build (release build) should build without errors

#### Actual behavior
```
--- stderr: dwb_plugins
In file included from /home/mkhansen/ros2_dev/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node.hpp:813:0,
                 from /home/mkhansen/ros2_dev/navigation2_ws/install/nav2_lifecycle/include/nav2_lifecycle/lifecycle_helper_interface.hpp:18,
                 from /home/mkhansen/ros2_dev/navigation2_ws/install/nav2_lifecycle/include/nav2_lifecycle/lifecycle_node.hpp:22,
                 from /home/mkhansen/ros2_dev/navigation2_ws/install/dwb_core/include/dwb_core/trajectory_generator.hpp:43,
                 from /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_plugins/include/dwb_plugins/standard_traj_generator.hpp:4 ,
                 from /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_plugins/include/dwb_plugins/limited_accel_generator.hpp:4 ,
                 from /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_plugins/src/limited_accel_generator.cpp:35:
/home/mkhansen/ros2_dev/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node_impl.hpp: In member function ‘virtual void dwb_plugins::LimitedAccelGenerator::checkUseDwaParam(const SharedPtr&)’:
/home/mkhansen/ros2_dev/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node_impl.hpp:286:21: error: ‘use_dwa’ is used uninitialized in this function [-Werror=uninitialized]
   rclcpp::Parameter param(name, parameter);
                     ^~~~~
/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_plugins/src/limited_accel_generator.cpp:69:8: note: ‘use_dwa’ was declared here
   bool use_dwa;
        ^~~~~~~
In file included from /home/mkhansen/ros2_dev/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node.hpp:813:0,
                 from /home/mkhansen/ros2_dev/navigation2_ws/install/nav2_lifecycle/include/nav2_lifecycle/lifecycle_helper_interface.hpp:18,
                 from /home/mkhansen/ros2_dev/navigation2_ws/install/nav2_lifecycle/include/nav2_lifecycle/lifecycle_node.hpp:22,
                 from /home/mkhansen/ros2_dev/navigation2_ws/install/dwb_core/include/dwb_core/trajectory_generator.hpp:43,
                 from /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_plugins/include/dwb_plugins/standard_traj_generator.hpp:4 ,
                 from /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_plugins/src/standard_traj_generator.cpp:35:
/home/mkhansen/ros2_dev/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node_impl.hpp: In member function ‘virtual void dwb_plugins::StandardTrajectoryGenerator::checkUseDwaParam(const SharedPtr&)’:
/home/mkhansen/ros2_dev/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node_impl.hpp:286:21: error: ‘use_dwa’ is used uninitialized in this function [-Werror=uninitialized]
   rclcpp::Parameter param(name, parameter);
                     ^~~~~
/home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_plugins/src/standard_traj_generator.cpp:93:8: note: ‘use_dwa’ was declared here
   bool use_dwa;
        ^~~~~~~
In file included from /home/mkhansen/ros2_dev/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node.hpp:813:0,
                 from /home/mkhansen/ros2_dev/navigation2_ws/install/nav2_lifecycle/include/nav2_lifecycle/lifecycle_helper_interface.hpp:18,
                 from /home/mkhansen/ros2_dev/navigation2_ws/install/nav2_lifecycle/include/nav2_lifecycle/lifecycle_node.hpp:22,
                 from /home/mkhansen/ros2_dev/navigation2_ws/install/dwb_core/include/dwb_core/trajectory_generator.hpp:43,
                 from /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_plugins/include/dwb_plugins/standard_traj_generator.hpp:4 ,
                 from /home/mkhansen/ros2_dev/navigation2_ws/src/navigation2/nav2_dwb_controller/dwb_plugins/src/standard_traj_generator.cpp:35:
/home/mkhansen/ros2_dev/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node_impl.hpp: In function ‘param_t nav_2d_utils::loadParameterWithDeprecation(const SharedPtr&, std::__cxx11::string, std::__cxx11::string, const param_t&) [with param_t = double]’:
/home/mkhansen/ros2_dev/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node_impl.hpp:286:21: error: ‘value’ is used uninitialized in this function [-Werror=uninitialized]
   rclcpp::Parameter param(name, parameter);
                     ^~~~~
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/standard_traj_generator.dir/src/limited_accel_generator.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
cc1plus: all warnings being treated as errors
make[2]: *** [CMakeFiles/standard_traj_generator.dir/src/standard_traj_generator.cpp.o] Error 1
make[1]: *** [CMakeFiles/standard_traj_generator.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [all] Error 2
---
Failed   <<< dwb_plugins        [ Exited with code 2 ]

Summary: 25 packages finished [12.0s]
  1 package failed: dwb_plugins
  1 package had stderr output: dwb_plugins
  3 packages not processed
```

#### Additional information

I changed the code in the offending files to this:
```
-  bool use_dwa;
-  nh->get_parameter("use_dwa", use_dwa);
+  bool use_dwa = false;
+  nh->get_parameter_or("use_dwa", use_dwa, false);
```
Then when I build I see an error at the rclcpp lifecycle implementation level:
```
/home/mkhansen/ros2_dev/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node_impl.hpp: In function ‘param_t nav_2d_utils::loadParameterWithDeprecation(const SharedPtr&, std::__cxx11::string, std::__cxx11::string, const param_t&) [with param_t = double]’:
/home/mkhansen/ros2_dev/ros2_ws/install/rclcpp_lifecycle/include/rclcpp_lifecycle/lifecycle_node_impl.hpp:286:21: error: ‘value’ is used uninitialized in this function [-Werror=uninitialized]
```
I'm not sure, is this a place where we need to `declare_parameter()` first?

bpwilcox, 2019-05-24:
I was looking at this issue on my build and tried a similar approach to initialize `use_dwa`. A problem with `declare_parameter()` here is that the parameter is already being declared in the `DWBLocalPlanner` node which I assume is being passed to the critics. However, at compile time, apparently that declaration is not visible.  We have this similar use case in nav2_costmap_2d, but perhaps the difference is because it is declared from a node in another package?

mkhansen-intel, 2019-05-31:
Fixed by #758

closed 2019-05-31 by mkhansen-intel