ros-navigation/navigation2 #757: Fix parameter declaration in Voxel and Obstacle plugins

orduno, 2019-05-24:
#752 fixes the parameter re-declaration issue with the static and inflation layer.

We might need to do something different for the Obstacle and Voxel plugins as these don't execute onInitialize() upon reset().

mkhansen-intel, 2019-05-31:
Does this refer to the lifecycle branch or the master branch? If it's lifecycle, can you add the label 'nav2_lifecycle' so we can track what issues exist in the new code vs. old?

yathartha3, 2019-11-01:
I am not seeing this issue right now after enabling the voxel layer. Open again if needed.

SteveMacenski, 2019-11-01:
Didn’t your PR have some obstacle layer namespaced parameters needed for the voxel layer?

What’s changed between your initial PR commit and now?

yathartha3, 2019-11-01:
@SteveMacenski Sorry for the confusion. I believe it was error on my part.

While working on my initial PR for voxel layer, I had removed the obstacle_layer parameters from the global_costmap namespace, and replaced it with voxel_layer params.
I __hadn't__ added the 'plugin_name' for global_costmap to have ["voxel_layer", ..]. So what was happening was that it was taking the default plugin_names, which included the 'obstacle_layer', and couldn't find those params.

Now in the param file, in both global and local costmap, I have specified the 'plugin_names'. Earlier, you would need params for 'obstacle_layer' and others which are in the default, even when it was not specifially defined in the params.

This is my understanding of what was happening.

closed 2019-11-01 by SteveMacenski