ros-navigation/navigation2 #768: Build broken with CMake < 3.13

crdelsey, 2019-05-29:
## Bug report

**Required Info:**

@rotu Found an issue with PR #658. Looks like older versions of CMake don't support one of the features used in that PR. See his comments for details.

#### Steps to reproduce issue
Build the stack with an older version of CMake.

#### Expected behavior
It builds.

#### Actual behavior
It doesn't.

#### Additional information

mkhansen-intel, 2019-05-31:
Let's clarify. Since we're supporting 18.04 and the default version of CMake for 18.04 is 3.10.2, we should support 3.10.2 and up for Crystal and Dashing.

https://packages.ubuntu.com/search?keywords=cmake

crdelsey, 2019-05-31:
That's interesting. 

I had assumed my machine had been updated from installing the regular Ubuntu patches, but I appear to still have CMake 3.10.2 installed. Yet I don't have this build failure.

There must be more to this.

rotu, 2019-05-31:
> That's interesting.
> 
> I had assumed my machine had been updated from installing the regular Ubuntu patches, but I appear to still have CMake 3.10.2 installed. Yet I don't have this build failure.
> 
> There must be more to this.

@crdelsey, which cmake is being invoked? You sure you're checking the right one?
I have two on my system:
via pip (3.13.3):
/usr/local/bin/cmake
via apt (3.10.2):
/usr/bin/cmake

crdelsey, 2019-05-31:
`/usr/bin/cmake` is the only one on my system. It's also the one used in CI, since the docker images are based on Ubuntu 18.04.

corsaircpt, 2019-06-14:
I ran into this as well https://github.com/ros-planning/navigation2/issues/856 . I have Dashing from sources and installed and Ubuntu 18.04 , the cmake is 3.10.2. Would appreciate help in getting this going.

crdelsey, 2019-06-14:
I've submitted PR #860 to address this problem. I haven't been able to reproduce the problem yet. @rotu  and @corsaircpt , I'd appreciate it if you could try out the PR and see if it fixes the problem for you.

rotu, 2019-06-14:
I can reproduce the issue reliably before this fix and can confirm the issue is resolved by your commit. Yay! Good job!

corsaircpt, 2019-06-14:
Hi, Applying PR #860 , fixes the issue for me , thank you for helping with this. Also compiles in turtlebot3 packages with Dashing in background.

crdelsey, 2019-06-17:
@corsaircpt @rotu Thanks for confirming this fixes the problem. 

I'm still concerned about the difference in behavior between your systems and ours. If you get a chance, could you check your colcon version to see if it matches?

Running `which colcon` produces `/home/crdelsey/.local/bin/colcon` and running `pip3 list | grep colcon` produces

```
colcon-argcomplete (0.3.2)
colcon-bash (0.3.2)
colcon-cmake (0.2.11)
colcon-common-extensions (0.2.0)
colcon-core (0.3.22)
colcon-defaults (0.2.2)
colcon-devtools (0.2.1)
colcon-library-path (0.2.1)
colcon-metadata (0.2.2)
colcon-notification (0.2.7)
colcon-output (0.2.3)
colcon-package-information (0.2.2)
colcon-package-selection (0.2.4)
colcon-parallel-executor (0.2.4)
colcon-pkg-config (0.1.0)
colcon-powershell (0.3.4)
colcon-python-setup-py (0.2.1)
colcon-recursive-crawl (0.2.0)
colcon-ros (0.3.10)
colcon-test-result (0.3.3)
colcon-zsh (0.3.2)
```

Please let me know if you get anything different.

rotu, 2019-06-17:
My colcon is installed globally, as I installed it through apt, as per [here](https://index.ros.org/doc/ros2/Installation/Dashing/Linux-Development-Setup/).

`which colcon`

> /usr/bin/colcon

`apt show python3-colcon-common-extensions`

> Package: python3-colcon-common-extensions
> Version: 0.2.0-2
> Priority: optional
> Section: python
> Maintainer: Dirk Thomas <web@dirk-thomas.net>
> Installed-Size: 21.5 kB
> Depends: python3:any (>= 3.5~), python3-colcon-argcomplete, python3-colcon-bash, python3-colcon-cmake, python3-colcon-core, python3-colcon-defaults, python3-colcon-devtools, python3-colcon-library-path, python3-colcon-metadata, python3-colcon-notification, python3-colcon-output, python3-colcon-package-information, python3-colcon-package-selection, python3-colcon-parallel-executor, python3-colcon-powershell, python3-colcon-python-setup-py, python3-colcon-recursive-crawl, python3-colcon-ros, python3-colcon-test-result, python3-colcon-zsh
> Download-Size: 2,770 B
> APT-Manual-Installed: yes
> APT-Sources: http://packages.ros.org/ros2/ubuntu bionic/main amd64 Packages
> Description: Meta package aggregating colcon-core and common extensions.
>  colcon-common-extensions
>  ========================
>  .
>  A meta package aggregating `colcon-core <https://github.com/colcon/colcon-core>`_ as well as a set of common extensions.

`pip3 list | grep colcon` gives almost the same output except for my `colcon-cmake`, which is version 0.2.9 (though I can't see anything between 0.2.9 and 0.2.11 which would explain the behavioral difference)

PR #860 (2019-06-19): Move install commands to subdirs where target is defined
## Basic Info

| Info | Please fill out this column |
| ------ | ----------- |
| Ticket(s) this addresses   | fixes #768, fixes #856  |
| Primary OS tested on | Ubuntu 18.04 |
| Robotic platform tested on | NA |

---

## Description of contribution in a few bullet points

Moves `install` commands to the subdirectory where the library is defined to properly support CMake 3.10.2 on some systems.
closed 2019-06-19 by crdelsey