ros-navigation/navigation2 #775: Commanded velocities are hardcoded

rotu, 2019-05-30:
The commanded speed in `nav2_motion_primitives::Spin::timedSpin` is hardcoded to 0.5 which is below the minimum turning velocity of my test robot. This value should probably be extracted to a parameter.

mkhansen-intel, 2019-05-31:
I agree this is a bug. We have a couple of other open issues to re-do the spin code, we'll need to fix this as well.

crdelsey, 2019-07-22:
Related to #423 and robot parameter server change

mhpanah, 2019-07-30:
#423 Did not address hardcoded commanded velocities.  Next PR will address this and also hardcoded yaw angle.

SteveMacenski, 2019-10-10:
@mhpanah @crdelsey anyone going to touch this? I can add to my queue. This has been in here a little longer than I'd like

SteveMacenski, 2019-10-11:
In progress

closed 2019-10-15 by SteveMacenski