ros-navigation/navigation2 #786: Migrate to Behavior Tree CoRoutines

crdelsey, 2019-05-31:
## Feature request

#### Feature description
PR #761 backs out the use of the CoRoutine feature in the behavior tree navigator due to some issues with losing incoming messages. We should root cause that problem and re-enable the coroutine feature.

#### Implementation considerations

mkhansen-intel, 2019-05-31:
I don't know the impact of this one, what if we never do it? Is something broken without Coroutines?

crdelsey, 2019-05-31:
Maybe no real impact, but it would make me feel better for 2 reasons:
1. Fewer threads means fewer race conditions,
2. There was a problem in our old coroutine implementation that isn't fully understood. What if it is still there in the thread based implementation, but happens less frequently?

crdelsey, 2019-06-10:
DUpe ofo #820

closed 2019-06-10 by crdelsey