ros-navigation/navigation2 #792: dwb consumes more than 150% cpu power when local planner is not running

mhpanah, 2019-05-31:
## Bug report

**Required Info:**

- Operating System:
  -Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  -lifecycle branch
- DDS implementation:
  - Fast-RTPS, 
 -ros2/src/eProsima/Fast-RTPS     -- “v1.7.2”
 -ros2/src/ros2/rmw_fastrtps       -- “0.7.1”

- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Run our stack and look at top

#### Expected behavior
If local planner is not actuating the robot, the dwb thread should not consume 150% cpu
#### Actual behavior
dwb thread is consuming lots of cpu
#### Additional information
[image]

----
## Feature request

#### Feature description

#### Implementation considerations

SteveMacenski, 2019-05-31:
Its probably the costmap rather than the dwb action if there's nothing being processed.

rotu, 2019-06-11:
Any info on what's causing this? I'm seeing this on a bunch of Nav2 nodes, even though they are all in unconfigured status:

[image]
[image]

crdelsey, 2019-06-11:
@mjeronimo found this block that seems to be consuming as much CPU as available.
https://github.com/ros-planning/navigation2/blob/c530cd89eb2e23cd4e9928c9df9dcdf21eaa042a/nav2_util/src/lifecycle_node.cpp#L53-L55

mjeronimo, 2019-06-20:
Fixed with #866.

closed 2019-06-20 by mjeronimo