ros-navigation/navigation2 #800: Cancelling nav2_goal causes the stack to crash

mhpanah, 2019-06-03:
## Bug report

**Required Info:**

- Operating System:
  -Ubuntu 18.04
- Installation type:
  - Source
- Version or commit hash:
  -lifecycle branch
- DDS implementation:
  - Fast-RTPS, 
 -ros2/src/eProsima/Fast-RTPS     -- “v1.7.2”
 -ros2/src/ros2/rmw_fastrtps       -- “0.7.1”

- Client library (if applicable):
  - rclcpp
#### Steps to reproduce issue

- Run stack, from rviz click on `Navigation2 Goal` 
- In the middle of the execution click on the `Cancel`
- Click on  `Navigation2 Goal` to set a new goal

#### Expected behavior
Start navigation to the new goal
#### Actual behavior
[image]

orduno, 2019-06-04:
I ran a test with #804 applied and am not seeing the stack crash. However, I do see another issue:

[image]

You'll notice after the second time I cancel the navigation, I provide the third goal and the nav stack immediately returns succeeded. I don't see any output from DWB.

mhpanah, 2019-06-05:
@orduno #804 might have fixed this issue.  I will test it out tomorrow.

mhpanah, 2019-06-07:
This issue is fixed in #804.

closed 2019-06-07 by mhpanah