ros-navigation/navigation2 #812: Re-enable lint checks for failing lifecycle branch packages

mkhansen-intel, 2019-06-07:
## Bug report
On the lifecycle branch, linters were failing for the following packages:
- [ ] nav2_bt_navigator
- [ ] nav2_costmap_2d 
- [ ] nav2_lifecycle_manager 
- [ ] nav2_mission_executor 
- [ ] nav2_tasks 
- [ ] nav2_robot

These are disabled by #783 to enable the CI to pass the system test. 
The lint errors need to be fixed and the tests need to be re-enabled.

mjeronimo, 2019-06-11:
@bpwilcox @orduno  nav2_bt_navigator and nav2_robot are already clean. I've submitted #839 which cleans up nav2_lifecycle_manager and nav2_tasks. If one of you can do nav2_costmap_2d and then re-enable the lint checks, we can finish this one.

mjeronimo, 2019-06-20:
Together, #839 and #840 address this issue.

closed 2019-06-20 by mjeronimo