ros-navigation/navigation2 #813: Clean up parameters for system test

mkhansen-intel, 2019-06-07:
## Bug report

The system test currently uses environment vars set in the CMakeLists.txt to set the paths to the files for maps, etc. There is also a nav2_params.yaml file which can also be used to set params. The current method is a hybrid of both. We should clean this up a bit and see if it can be made simpler.

mkhansen-intel, 2019-06-20:
Adding to this - now that we can successfully use 'use_sim_time' in the params file for all nodes and sub-nodes, we shouldn't need to set it in the system test anymore. We should be able to use the nav2_bringup_launch.py launch file and nav2_params.yaml files from the nav2_bringup package instead of custom copies in the nav2_system_tests package.

mkhansen-intel, 2019-06-20:
check that - we still need a custom launch file to add the transforms, but it could include the other existing launch file and params file.

orduno, 2019-10-21:
@mkhansen-intel Was this addressed by #1243? What are we missing?

mkhansen-intel, 2019-10-25:
Yes, #1243 closed this. I had previously removed the "setSimTime" code.

closed 2019-10-25 by mkhansen-intel