ros-navigation/navigation2 #827: Change nav2_bringup launch script to launch gzserver and gzclient instead of calling gazebo

crdelsey, 2019-06-10:
Sometimes gazebo crashes on startup, leaving the gzserver sub-process running in the background. When the launch is restarted, the existing gzserver process causes gazebo to crash again.

To work around this, change the launch file to launch gzserver and gzclient independently. That way the launch tool manages the subprocesses. It should terminate them both if the launch script is ended for any reason.

mkhansen-intel, 2019-06-20:
I'm not going to have time to get to this one before release.

SteveMacenski, 2019-11-06:
I'm curious, does anyone _want_ the front end? Its super heavy and really gets in the way for me. 

I'd vote to just launch only `gzserver` and make launching `gzclient` up to the user if they care

mlherd, 2019-11-09:
Yes, it is heavy, but I almost always use gzclient. Maybe the launch file can take an argument such as use_gazebo_client:=True ?

mlherd, 2019-11-11:
```
    declare_simulator_cmd = DeclareLaunchArgument(
        'simulator',
        default_value='gazebo',
        description='The simulator to use (gazebo or gzserver)')
```
Currently, if the argument ```simulator``` is set to ```gzserver``` it will not launch ```gzclient```. The default value for ```simulator``` is ```gazebo``` that is why it launches gzserver and gzclient together.

> change the launch file to launch gzserver and gzclient independently
I will work on this issue first and then see how I can make it more convenient for the user to enable/disable gzclient.

SteveMacenski, 2019-11-11:
The generic phrase we use in simulation is `headless` as a bool to start the client. This has history in ROS1 as well.

closed 2019-11-18 by mlherd