ros-navigation/navigation2 #833: Change hard-coded urdf model in launch file to launch parameter

bpwilcox, 2019-06-10:
The `nav2_bringup_launch.py` has the urdf model hard-coded using the `turtlebot3_burger.urdf` in the `robot_state_publisher`. This should be a parameter instead.

mkhansen-intel, 2019-06-20:
The new nav2_bringup_launch.py in #841 fixes this. The robot_state_publisher is expected to be run separately, and is not launched in nav2_bringup_launch.py

This hard-coding still exists in the nav2_simulation_launch.py however.

mkhansen-intel, 2019-06-25:
@bpwilcox - I'm going to close this, since it is no longer an issue for the nav2_bringup_launch.py file. It IS still part of the simulation launch file. If we want to change that one, either re-open and clarify or open a new issue. I would say at least that is a lower priority than this.

closed 2019-06-26 by mkhansen-intel