ros-navigation/navigation2 #855: BT-enable the lifecycle manager

mjeronimo, 2019-06-13:
## Feature request

#### Feature description
The current lifecycle manager interfaces to the nav2 lifecycle nodes via the lifecycle services that the nodes present. The lifecycle manager accepts a parameter that specifies a list of node names that it is to manage. The intention was to keep the lifecycle manager generic so that it could manage various nodes without recoding. However, the current situation results in a fairly limited lifecycle manager. It is simply able to walk through the node list and configure and activate each node in turn. Because of the desire to keep it generic, it does not perform any more detailed checks such as checking that data is flowing on specific topics. 

The feature suggestion is to enable the Lifecycle Manager with Behavior Trees. The Lifecycle Manager would then read an external XML file which specifies the bringup (and shutdown) behavior. There can be BT nodes that do things like start up a node (configure and activate) and shut down a node (deactivate, cleanup, and shutdown). Various other BT node types are possible that would be helpful in a startup or shutdown situation. 

As a general lifecycle manager, the module could perhaps execute other non-startup/shutdown BTs. Perhaps, bringing all of the nodes to a lifecycle state that permits reconfiguration (re-reading node parameters and configuring buffers, for example). 

It remains to be seen whether some of this functionality should be part of the ROS2 launch system. The current ROS2 launch system seems to be oriented around launch nodes only. Although one can interact with lifecycle nodes to some degree, the launch system does not seem to be designed to be a general lifecycle manager. So, the though is to use the launch script to start the nodes and then rely on the lifecycle manager to control the nodes (control startup and shutdown sequences at least). 

[BT-Enabled Lifecycle Manager.pdf](https://github.com/ros-planning/navigation2/files/3288213/BT-Enabled.Lifecycle.Manager.pdf)

crdelsey, 2019-07-08:
Should this be done for the e release?

SteveMacenski, 2020-03-02:
Per #1472 and other discussions over the last year, I think we've come to the agreement to push this into the launch system. We'll keep lifecycle manager around (spun out into another repo) as its generally useful, but we'll have the launch system handle this.

closed 2020-03-02 by SteveMacenski