ros-navigation/navigation2 #861: Handle a pending preemption after reaching a final state on current goal.

orduno, 2019-06-16:
## Feature request

#### Feature description
Define what to do with a pending preemption before setting a final state (succeed / cancel / abort) based on the current goal.

#### Implementation considerations

For example, in the case of DWB (path following) if the end pose is different between the current pose and the end pose we might decide to pursue the new goal, however, if these have the same end pose we might set the final state without preempting.

SteveMacenski, 2019-10-24:
@crdelsey I see a TODO in the nav2 controller on implementing the preemption condition. Is this tied in to that?

Any particular blocker?

orduno, 2019-12-09:
Solved in #1356.

closed 2019-12-09 by orduno