ros-navigation/navigation2 #864: nav2_map_server can not load map in ROS2 dashing

ZakariaChekakta, 2019-06-18:
## Bug report
nav2_map_server can not load map in ROS2 dashing
**Required Info:**

- Operating System:
Ubuntu 18.04, ROS2 dashing
- Installation type:
 source  
- Version or commit hash:
  - 
- DDS implementation:
default
- Client library (if applicable):
rclcpp

#### Steps to reproduce issue

 As we follow https://github.com/ros-planning/navigation2/tree/master/nav2_map_server to load map by ros2 navigation, the map not uplaod when we run

ros2 run nav2_map_server map_server __params:=map_server_params.yaml

```

#### Expected behavior
Can display map in RVIZ2.
#### Actual behavior
Unable to display map in RVIZ2.
#### Additional information

We  show topics in:
$ros2 topic list
/clock
/map_server/transition_event
/parameter_events
/rosout
/tf

$ ros2 node list
/map_server

crdelsey, 2019-06-19:
There are two problems I expect you are having:

1. map_server is now a lifecycle node and needs to be transitioned to the active state. There are several ways you can trigger that state transition. The easiest ways from the command line are
```
ros2 run nav2_util lifecycle_bringup map_server
```
or 
```
ros2 lifecycle set map_server configure
ros2 lifecycle set map_server activate
```

2. PR #863, which was integrated yesterday helps improve communication reliability between map_server and RViz2. You'll want to make sure you are using a version that includes that PR

ZakariaChekakta, 2019-06-21:
@crdelsey Thank u so much Sr, It works now

crdelsey, 2019-06-24:
@ZakariaChekakta I'm glad that this fixed your problem. Please file issues for any other problems you run into.

closed 2019-06-24 by crdelsey