ros-navigation/navigation2 #867: Map not showing in RVIZ2 when running with autostart = True

mkhansen-intel, 2019-06-18:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - master branch as of 6/18
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

Launch RVIZ2
Launch the stack with autostart = True
(see nav2 bringup instructions in #841)

#### Expected behavior
Map should appear in RVIZ2 when the map_server node gets to the 'active' lifecycle state

#### Actual behavior
The map doesn't appear. This may be due to the RVIZ QOS settings being default, and the map_server publishing with 'transient local' and 'reliable'. 

#### Additional information
For now we work around this with #863 - republishing the map every 2 seconds.

mkhansen-intel, 2019-06-20:
I believe this is related to: https://github.com/ros2/rviz/issues/400

Rviz doesn't currently support the transient local mode, which is why we see this issue.

mkhansen-intel, 2019-09-17:
Adding link to #331 - that needs to be removed when this is fixed.

SteveMacenski, 2019-10-15:
https://github.com/ros-planning/navigation2/pull/1199 lists this ticket as resolved

closed 2019-10-15 by SteveMacenski