ros-navigation/navigation2 #870: bt_navigator crashes when NavigateToPose is called before the node is active

rotu, 2019-06-19:
I'm observing crashes of bt_navigator when receiving a `NavigateToPose` request before the node has been completely brought up. I added the following code to `BtNavigator::navigateToPose` and observed that `tree_` is not yet initialized `if (!tree_) {RCLCPP_ERROR(get_logger(), "Tree is null");}`. I believe this is responsible for the crash later on in the call `bt_->run(tree_, on_loop, is_canceling);`

BtNavigator shouldn't be trying to execute for `NavigateToPose` actions until `on_activate` has been called.

```
[lifecycle_manager-7] [INFO] [lifecycle_manager]: Configuring and activating bt_navigator
[bt_navigator-5] [INFO] [bt_navigator]: Configuring
[bt_navigator-5] [INFO] [bt_navigator_client_node]: Waiting for "ComputePathToPose" action server
[bt_navigator-5] [INFO] [bt_navigator_client_node]: "ComputePathToPose" BtActionNode initialized
[bt_navigator-5] [INFO] [bt_navigator_client_node]: Waiting for "FollowPath" action server
[bt_navigator-5] [INFO] [bt_navigator_client_node]: "FollowPath" BtActionNode initialized
[bt_navigator-5] [INFO] [bt_navigator_client_node]: Waiting for "Spin" action server
[amcl-2] [INFO] [amcl]: initialPoseReceived
[amcl-2] [WARN] [amcl]: Failed to transform initial pose in time (Lookup would require extrapolation into the future.  Requested time 1560901938.82616 but the latest data is at time 1560901938.82288, when looking up transform from frame [base_footprint] to frame [odom])
[amcl-2] [INFO] [amcl]: Setting pose (1560901938.826306): 0.152 -0.318 -2.529
[amcl-2] [INFO] [amcl]: initialPoseReceived
[amcl-2] [INFO] [amcl]: Setting pose (1560901953.633383): -1.033 0.550 2.606
[amcl-2] [WARN] [amcl]: Failed to transform initial pose in time (Lookup would require extrapolation into the future.  Requested time 1560901953.63328 but the latest data is at time 1560901953.61795, when looking up transform from frame [base_footprint] to frame [odom])
[bt_navigator-5] [INFO] [bt_navigator]: Begin navigating from current location to (0.31, 0.68)
[bt_navigator-5] [ERROR] [bt_navigator]: Tree is null
[ERROR] [bt_navigator-5]: process has died [pid 12628, exit code -11, cmd '/home/snuc/ros2_ws/install/lib/nav2_bt_navigator/bt_navigator __node:=bt_navigator __params:=/home/snuc/ros2_ws/install/share/openrover_demo/config/nav2.yaml'].
```

mjeronimo, 2019-06-19:
@orduno 

@rotu How is it that the NavigateToPose request is issued before the stack is up? Was this simply specifying a goal pose before the stack was completely up and running?

Actions, services, and subscribers are active once they have been created. Each of these may need their own solutions since ROS doesn't provide any lifecycle-style support here. For example, our SimpleActionServer template can ensure that our action servers are in the Active state before accepting any actions to execute. Services also should check that the node that they are providing services for is Active before executing the request (fail otherwise). Same for subscriptions. We're currently doing this in AMCL, for example, where we ensure that we're active before bothering to process incoming laser scan messages.

rotu, 2019-06-19:
Yes, this was specifying a goal pose in RViz before the nav2 stack was completely up. I was trying to use the `navigate_w_replanning_and_recovery.xml` behavior tree file. Bringup was hung for at least 10 seconds (and probably indefinitely) waiting for the "Spin" action server to be ready.

orduno, 2019-06-19:
#880 fixed the issue for me. @rotu Can you give it a try?

rotu, 2019-06-20:
@orduno that seems to have done the trick!

rotu, 2019-06-20:
oops. reopening - the PR isn't merged in yet :-p

orduno, 2019-06-25:
Got fix #880 merged, closing issue.

closed 2019-06-25 by orduno