ros-navigation/navigation2 #881: Cannot see error messages from map_server

crdelsey, 2019-06-19:
## Bug report

When running map_server with an invalid map file or filename, all useful error messages from exceptions are hidden. All I get is:
```
[INFO] [map_server]: Creating
[INFO] [map_server]: Configuring
[WARN] []: Error occurred while doing error handling.
[FATAL] [map_server]: Lifecycle node entered error state
```

This is likely due to the lifecycle transition service call catching the exception and silently transitioning to the error state, like we've seen in other nodes.

rotu, 2019-07-01:
Blech. I just encountered this. Any fixes/workarounds?

crdelsey, 2019-07-02:
Uncomment these lines
https://github.com/ros2/rclcpp/blob/378657865e61b9a0b0a1775d7391dcb4a950cf27/rclcpp_lifecycle/src/lifecycle_node_interface_impl.hpp#L429-L435

@yathartha3 Was working on a PR for this. Any update on whether that is ready?

rotu, 2019-07-03:
Thanks, @crdelsey. Can confirm that patched the issue. Created bug report in rclcpp repo.

yathartha3, 2019-07-06:
Seems to be working on after un-commenting those lines. I submitted a PR for in the rclcpp repo.

closed 2019-07-06 by yathartha3

crdelsey, 2019-07-08:
Upstream PR is ros2/rclcpp#777