ros-navigation/navigation2 #887: Replace time-based global path replanning with a condition-based approach

mjeronimo, 2019-06-24:
## Feature request

Currently, the navigation Behavior Trees implement continuous path replanning using a time-based approach. That is, every n seconds, a new path is computed and made available to the local planner, who receives the path and updates its goal accordingly. This approach isn't very sophisticated and can result in unnecessary work (perhaps there are no new obstacles and the previous path was entirely sufficient). Instead, as Carlos has suggested in #807, we can use a condition-based approach where we are smarter about when we request a new global path recomputation. For example, the map has changed, there are obstacles in the path, etc. This could be implemented as a condition node in the Behavior Tree, such as a ShouldRecomputePath condition, which can then control when a new path is computed.

crdelsey, 2019-07-08:
One problem this would address is constantly changing (flip-flopping) paths from the global planner. Another alternative would be to add a filter to decide if the recomputed plan is worth passing on to the controller.

SteveMacenski, 2019-07-11:
I agree with @crdelsey, I think we discussed this a couple months ago. I think making it condition based on top of frequency based would be nice, but complex. As you approach a goal or you see more new areas of your environment, you're going to want to replan often to deal with the differences of the state of your space and the perfect map you have built that is never actually perfect in reality. Things move and frequently replanning is good.

I think I've brought up before the idea of path QA, in which we have an intermediate program checking the difference in the paths and decide if we should change or keep on the same one. 

I think this falls into the K.I.S.S. principle. Having this run on the robot edge versus the cloud changes the dynamics of what I consider unnesessary work. Either way your CPU is spinning around and if once a second there's an extra task, it's not going to be the difference of putting it into the power saving mode -- so there's no benefit of running your robot's processor any lower than 80% on the edge when you could be doing something even marginally useful.

crdelsey, 2019-10-02:
Closing till we've got a specific problem to address with this change.

closed 2019-10-02 by crdelsey