ros-navigation/navigation2 #888: New Behavior Tree Control Nodes in C++ versus XML equivalents

mjeronimo, 2019-06-24:
In some cases, there are custom Behavior Tree constructs that we might find desirable for navigation. For example, Mohammad has proposed a RecoveryNode that executes the first child (the 'action branch'), and if that fails, executes a second child (the 'recovery branch'), then repeats that sequence a specific number of times (that's a quick description to get the idea, not a precise specification). 

With nodes like this, we have a couple choices for implementation:

1. Compose the functionality from existing nodes

A positive for this approach is that users don't have new control nodes to learn. Drawbacks including a more complex BT to understand and the potential duplication of a few nodes in several places in a BT (every place that the recovery behavior is required). Since the BT library is using tinyXML2, we don't have XML includes at run-time, which could be used to compose the BTs from parts and therefore eliminate duplication. A potential work-around is that we could build the BTs at build-time and use a tool that does support XML includes. This way, a sub-tree, like a Recovery sub-tree, could be specified once, but used in multiple places.

2. Write a custom node in C++

This has the advantage of avoiding the duplication of the XML-based approach. Drawback is that its a new node type for users. However, this node type should be pretty intuitive and easy to understand. One possibility is to push this node type into the BT library itself.

mhpanah, 2019-06-24:
Generally, I prefer composing the functionality from existing control nodes over a custom C++ node.  However, for [recovery node](https://github.com/ros-planning/navigation2/issues/832), since I believe this is a fundamental control node that is needed beyond navigation, I prefer writing a custom node in C++ for the recovery node and pushing it upstream to the BT library.   The idea of Recovery node is not new and it has previously been used in literature beyond navigation. 

As a general rule of thumb, I think a user of the BT library should be able to define any custom nodes for conditions and actions but for the control node, one should consider composing the functionality from existing control nodes. If there is a need for implementing a new custom node in C++ then that node should be pushed upstream to the BT library. If the broader community rejects the proposed C++ custom node then it is preferable to implement that node by composing existing nodes.

facontidavide, 2019-06-25:
Actually I believe the opposite than @mhpanah . I would suggest to create your own C++ node.

As developer of the BT library, I am in favor of seeing as a Domain Specific language that the user can extend.

But to do a fair comparison, can you show me how you would implement this logic using the existing built-in nodes such as RepeatUntilSuccesfull and Fallback?

mjeronimo, 2019-06-25:
@facontidavide Regarding the Domain Specific Language concept... I think that there are certainly conditions, actions, and decorators that are domain specific. I'm wondering, however, about the control nodes. In some cases, such as the RecoveryNode idea, seem to be general control constructs and not domain specific. I suppose there could be domain-specific control nodes as well. If control nodes are generally useful, they may be candidates for the core library and not the user's extensions. What do you think?

facontidavide, 2019-06-25:
Just send a PR to the main library and if it make sense I will happily merge it

mjeronimo, 2019-06-25:
@facontidavide Sure, thanks for the input.

mjeronimo, 2019-06-25:
Closing the issue with the following summary:

The approach we will take will be to create our own domain-specific node types in C++, as needed. This may include control nodes. If we think that a node type is generally useful, we can submit a PR upstream to the BT library.

closed 2019-06-25 by mjeronimo