ros-navigation/navigation2 #893: Build failed (test_msgs in nav2_util)

routiful, 2019-06-26:
## Bug report
- Operating System:
  - Linux Mint 19.1
- Installation type:
  - source
- Version or commit hash:
  - master branch
- DDS implementation:
  - Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
$ colcon build --symlink-install
```

#### Expected behavior

#### Actual behavior
```
Starting >>> nav2_util
Starting >>> nav_2d_utils
--- stderr: nav2_util                                                                         
CMake Error at CMakeLists.txt:16 (find_package):
  By not providing "Findtest_msgs.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "test_msgs", but CMake did not find one.

  Could not find a package configuration file provided by "test_msgs" with
  any of the following names:

    test_msgsConfig.cmake
    test_msgs-config.cmake

  Add the installation prefix of "test_msgs" to CMAKE_PREFIX_PATH or set
  "test_msgs_DIR" to a directory containing one of the above files.  If
  "test_msgs" provides a separate development package or SDK, be sure it has
  been installed.

---
Failed   <<< nav2_util	[ Exited with code 1 ]
Aborted  <<< nav_2d_utils  
```
#### Additional information

Build failed at nav2_util

crdelsey, 2019-06-26:
This error suggests you don't have the test_msgs package installed. How did you install ROS2? Did you build it from source or did you install the Dashing packages?

If you built ROS2 from source, do you have the `ros2/rcl_interfaces/test_msgs` package in your ROS 2 workspace and did it build successfully?

If you installed the Dashing packages, does rosdep fix the problem for you?
```bash
> source /opt/ros/dashing/setup.bash
> rosdep install --ignore-src --from-paths /path/to/navigation2
```
If you are using the packages and rosdep doesn't work for you, please post any error messages from rosdep. If rosdep doesn't work, a workaround would be to install the package manually
```
sudo apt install ros-dashing-test-msgs`
```

routiful, 2019-06-27:
@crdelsey 
Thank you for quick reply.
I have been installed Dashing by Binary packages.

I have solved to install itand tf2-sensor-msgs by `apt` command 

```
sudo apt install ros-dashing-test-msgs
sudo apt install ros-dashing-tf2-sensor-msgs
```

closed 2019-06-27 by routiful