ros-navigation/navigation2 #901: ROS2 Navigation2 simulate cannot initial pose and Rviz2 will force close

rafaelhuang-intel, 2019-07-01:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source 
- Version or commit hash:
  - master - 43f8e726b9bab2eda5767203d5b30ab64c1ad69d
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
 1. Run simulate command:

termianl 1
$ export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:/home/intel/robot_devkit/rdk_ws/Turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/models

$ gazebo --verbose -s libgazebo_ros_init.so /home/intel/Desktop/new_map/turtlebot3_ros2_demo.world

termina 2
$ source /opt/robot_devkit/robot_devkit_setup.bash
$ ros2 launch turtlebot3_bringup turtlebot3_state_publisher.launch.py use_sim_time:=True

terminal 3
$ source /opt/robot_devkit/robot_devkit_setup.bash
$ ros2 launch nav2_bringup nav2_bringup_launch.py use_sim_time:=True autostart:=True map:=/home/intel/Desktop/new_map/map.yaml

terminal 4
$ source /opt/robot_devkit/robot_devkit_setup.bash
$ ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/launch/nav2_default_view.rviz

2. click the "2D Pose Estimate" button in RVIz2

3.  Check robot pose in Rviz2
```

#### Expected behavior
robot can set pose to expect position.
#### Actual behavior
robot can not set pose to expect position and Rviz2 will crash.
#### Additional information
[image]
[image]
[image]
[image]
[image]

----
## Feature request

#### Feature description

#### Implementation considerations

mkhansen-intel, 2019-07-01:
Looking at the picture, it doesn't look like you pressed 2D pose estimate, it looks like you pushed "Navigation2 Goal".  Is that what happened? 

Even if that's true, I do think this is a bug. We need to be able to handle this case gracefully.

rafaelhuang-intel, 2019-07-02:
@mkhansen-intel as I checked, if set initial pose by  press "2D pose estimate", the terminal also popup message 
[INFO] [rviz2]: Setting goal: Frame:map, Position(-1.98198, -0.456825, 0), Orientation(0, 0, -0.0207147, 0.999785) = Angle: -0.0414324
[INFO] [rviz2]: Setting goal: Frame:map, Position(-2.00427, -0.449416, 0), Orientation(0, 0, 0.048309, 0.998832) = Angle: 0.0966557
the robot module or TF not set to pose of you want, and than if you press "Navigation2 Goal", the Rviz2 will hang and cannot do anything. if wait a moment, Rviz2 will popup force quit message.

testkit, 2019-07-03:
Similar issue was found in rviz2 for real chassis (#905), but this rviz2 issue might be the curlprit ros2/rviz#417.

mlherd, 2019-07-08:
@rafaelhuang-intel

The error message "failed to load plug ..." usually means that you didn't source the environment setup files (setup.bash files) or the paths to the setup.bash files are wrong. In other words, it happens if the environment is not setup correctly before run Gazebo. I was able to generate the same error messages when I didn't source any setup.bash files (you can see them below) before I run Gazebo. I sourced them before launching any other launch files. When I do this, Gazebo, Rviz, Navigation launches but Navigation2Gaol command will get Rviz crashed or frozen.

**Solution:**
If you installed ROS from apt on Ubuntu then you will need to source setup.bash files in '/opt/ros/dashing/install'

If you are using ~/.bashrc file to source them, it should look like this. In your case the paths will look different since you installed the packages in different directories.

```
export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/ros2_all_ws/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/models
export TURTLEBOT3_MODEL=waffle

source ~/ros2_all_ws/ros2_ws/install/setup.bash
source ~/ros2_all_ws/turtlebot3_ws/install/setup.bash
source ~/ros2_all_ws/navstack_dependencies_ws/install/setup.bash
source ~/ros2_all_ws/navigation2_ws/install/setup.bash
```

If sourcing doesn't help, you can try to individually build the packages from source. You can find the instructions [here](https://github.com/ros-planning/navigation2/pull/896).

I will keep investigating this issue.

testkit, 2019-07-09:
@mlherd, thanks for your information, we'll try your suggestion. BTW, did you ever get nav2 run in Gazebo and rviz2 successfully?

rafaelhuang-intel, 2019-07-09:
@mlherd 

As I checked, all source already set to test environment, but Rviz2 still crashed or frozen. 
and the Rviz2 terminal display error message
"[ERROR] [rviz2]: "map" passed to lookupTransform argument target_frame does not exist. "

[image]
[image]
[image]
[image]
[image]

mlherd, 2019-07-09:
> @mlherd, thanks for your information, we'll try your suggestion. BTW, did you ever get nav2 run in Gazebo and rviz2 successfully?

 Yes, I did. You can see it working in the videos at the link below. I actually recorded more videos using TB3 Burger in Gazibo and Rviz today. I will upload them tomorrow when I get a chance. I haven't tested the navigation stack with a real tb3 yet, but I am working on it.

https://github.com/mlherd/navigation2/tree/new-tutorials/doc/tb3_tutorial

rafaelhuang-intel, 2019-07-09:
@mlherd

 we tried to apply rviz patch (https://github.com/Martin-Idel/rviz/tree/bugfix/map_display) from https://github.com/ros-planning/navigation2/issues/908, but issue still exist.

mlherd, 2019-07-09:
> @mlherd
> 
> As I checked, all source already set to test environment, but Rviz2 still crashed or frozen.
> and the Rviz2 terminal display error message
> "[ERROR] [rviz2]: "map" passed to lookupTransform argument target_frame does not exist. "
> 
> [image]
> [image]
> [image]
> [image]
> [image]

OK. At least it seems like the Gazebo plugin error massages are gone now.

I have seen this Rviz error messages before, but I don't remember how I fixed it. I will try to regenerate the same error message and let you know the possible solution. 

The "drop message frame base scan ..." error is something we have been observing. It should not cause a crash.

mlherd, 2019-07-09:
> @mlherd
> 
> we tried to apply rviz patch (https://github.com/Martin-Idel/rviz/tree/bugfix/map_display) from #908, but issue still exist.

I think the problem they were having is different than yours since they were trying to use a different robot than tb3 using their own map.

rafaelhuang-intel, 2019-07-09:
@mlherd 

I saw your tutorial (https://github.com/mlherd/navigation2/tree/new-tutorials/doc/tb3_tutorial), but I have a question that I see more dependencies (rosdep install --from-paths src --ignore-src --rosdistro crystal -y --skip-keys "console_bridge fastcdr fastrtps libopensplice67 libopensplice69 rti-connext-dds-5.3.1 urdfdom_headers"
)  us Crystal version.
I hope to know which version of ros2 core, dependence, Navigation2 you us it? all Dashing? or mixed of Dashing and Crystal?

mlherd, 2019-07-09:
@rafaelhuang-intel 

It is all Dashing. It was a mistake in the tutorial. I just fixed it. Sorry for the confusion.

rafaelhuang-intel, 2019-07-09:
@crdelsey @mlherd 
Carl, as I remember, currently  ROS2 Navigation2 Dashing release used dashing-devel branch, not master right?  so which different of master and dashing-devel?

crdelsey, 2019-07-09:
@rafaelhuang-intel You should use the dashing-devel branch when you are using the ROS 2 dashing release. You should use our master branch when using the ROS 2 master branch. 

However, right now it doesn't matter. We only have a couple of bug fixes on master, and I don't think they would fix this problem.

mlherd, 2019-07-10:
@rafaelhuang-intel I just submitted a PR for this issue. It should prevent Rviz crashing after setting a goal pose. I will keep working on the issue.

rafaelhuang-intel, 2019-07-10:
@mlherd 

After follow your tutorial (https://github.com/mlherd/navigation2/tree/new-tutorials/doc/tb3_tutorial) to build new test environment, I am using the master branch for ros2 and Navigation2, it's workable, but still have few time will meet this issue, if I reboot system and try again, issue disappear.

Rviz terminal still popup error message "[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 584.80900 but the latest data is at time 584.76600, when looking up transform from frame [base_scan] to frame [map]" in testing.

Navigation terminal will popup error message
"[lifecycle_manager-1] [ERROR] [lifecycle_manager]: Failed to change state for node: map_server
[lifecycle_manager-1] [ERROR] [lifecycle_manager]: Failed to bring up node: map_server, aborting bringup
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 32.201 for reason(0)"

BTW, I'm not find "ros-dashing-geometry2" package, checked github(https://github.com/ros/geometry2), not ROS2 version, I only installed "ros-dashing-geometry-msgs" & "ros-dashing-geometry-msgs-dbgsym" . So I hope to know which problem will meet it if no this package.

mlherd, 2019-07-10:
@rafaelhuang-intel 

I am glad to hear that the tutorial was helpful to you. Don't worry about ros-dashing-geomerty2. It doesn't exist in ROS2 Dashing. I changed it in the tutorial as well. 
You can use `sudo apt-get install ros-dashing-geometry*` instead. 

**Error messages:**
RVIZ: I get see the same error messages. However, the robot should still navigate. I don't know what causes these error messages yet.

**Navigation:** 
[world_model-4] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 32.201 for reason(0) should not cause a problem.

I don't see the other two messages. Are you sure the path to turtlebot3_world.yaml is correct? If there is no map map_server will not work.

[lifecycle_manager-1] [ERROR] [lifecycle_manager]: Failed to change state for node: map_server
[lifecycle_manager-1] [ERROR] [lifecycle_manager]: Failed to bring up node: map_server, aborting bringup

rafaelhuang-intel, 2019-07-11:
@mlherd 

I'm sure the path to turtlebot3_world.yaml is correct, because I meet this error message while in navigate.

I also meet other problem that if set goal from A to B, navigate will auto stop and robot not arrive expect point, but I can set goal again and robot workable except hit the post or wall, this issue meet rate ~30% ~ 40%, so do you meet same problem?

rafaelhuang-intel, 2019-07-11:
@mlherd 

Attached log, you can see above error messages.  

BTW, if robot hit the post in navigate, have few rate navigate will not auto cancel, in this case Rviz2 will crash. 

[error_message_log.txt](https://github.com/ros-planning/navigation2/files/3380040/error_message_log.txt)

mlherd, 2019-07-11:
@rafaelhuang-intel 

Since the initial issue (RVIZ crashes when setting initial pose) is solved, I will close this one.

Can you open new issues for the other problems.

Issue 1- "I also meet other problem that if set goal from A to B, navigate will auto stop and robot not arrive expect point, but I can set goal again and robot workable except hit the post or wall, this issue meet rate ~30% ~ 40%, so do you meet same problem?"

Issue 2- "I meet this error message while in navigate" and the log file.

I will be happy to work together with you on these issues.

Melih

rafaelhuang-intel, 2019-07-11:
@mlherd 

sure, I will open 2 new issue to trick these 2 issues and close this one.

closed 2019-07-11 by mlherd