ros-navigation/navigation2 #903: dwb_controller without the static layer causes a partial costmap

rotu, 2019-07-01:
dwb_controller relies on the static_layer plugin to set the map bounds. So when I exclude this plugin, the local costmap comes out truncated (see RViz screenshot). Also, dwb_controller spews out warnings. This problem goes away when adding the `nav2_costmap_2d::StaticLayer` plugin, even with `static_layer.enabled: False`.

[image]

Local costmap settings:
```
local_costmap:
  dwb_controller:
    ros__parameters:
      plugin_names: ["obstacle_layer", "inflation_layer"]
      plugin_types: ["nav2_costmap_2d::ObstacleLayer", "nav2_costmap_2d::InflationLayer"]
      global_frame: map
      always_send_full_costmap: true
      use_sim_time: false
      robot_radius: 0.17
      inflation_layer:
        cost_scaling_factor: 3.0
      obstacle_layer:
        enabled: True
      observation_sources: scan
      scan:
        topic: /scan
        max_obstacle_height: 2.0
        clearing: True
        marking: True
```

Logging warnings:
```[dwb_controller-4] [WARN] [local_costmap.dwb_controller]: Sensor origin at (-0.05, 2.29) is out of map bounds. The costmap cannot raytrace for it.
[dwb_controller-4] [WARN] [local_costmap.dwb_controller]: Sensor origin at (-0.05, 2.29) is out of map bounds. The costmap cannot raytrace for it.
[dwb_controller-4] [WARN] [local_costmap.dwb_controller]: Sensor origin at (-0.05, 2.29) is out of map bounds. The costmap cannot raytrace for it.
[dwb_controller-4] [WARN] [local_costmap.dwb_controller]: Sensor origin at (-0.05, 2.29) is out of map bounds. The costmap cannot raytrace for it.
[dwb_controller-4] [WARN] [local_costmap.dwb_controller]: Sensor origin at (-0.05, 2.29) is out of map bounds. The costmap cannot raytrace for it.
[dwb_controller-4] [WARN] [local_costmap.dwb_controller]: Sensor origin at (-0.05, 2.29) is out of map bounds. The costmap cannot raytrace for it.
```

crdelsey, 2019-07-02:
I don't see the `rolling_window: true` param in your params file.  This also seems to set map bounds, if I'm not mistaken. Maybe we need to spit out an error message that explains one or the other of these needs to be set.

rotu, 2019-07-03:
Thanks, @crdelsey. Adding `rolling_window: true` to my params did make the map stop truncating as well. An error message would be nice. Nicer yet would be more obvious interaction between plugins - I would naively expect that either the costmap has either fixed bounds set in parameters or it dynamically expands to fit all its layers.

crdelsey, 2019-07-08:
I noticed that in ROS 1, there was a `static_map` parameter (probably for move_base) that we seem to have lost. This seems to be the way the static layer was turned on and off in ROS 1. Maybe there was logic associated with this param in ROS 1 to make sure costmap dimensions were set.

bpwilcox, 2019-07-09:
@crdelsey the `static_map` parameter was used for backwards compatability with the "pre-hydro parameter style" I believe

bpwilcox, 2019-07-10:
@rotu In ROS1 navigation, there are parameters under the costmap node that resize the map bounds:  `map_width_meters`, `map_height_meters', `resolution`, `origin_x`, `origin_y` during dynamic reconfigure callbacks. We still have those parameters, but they are not not being declared and only [resize the map once during the configure step](https://github.com/ros-planning/navigation2/blob/master/nav2_costmap_2d/src/costmap_2d_ros.cpp#L108) of the lifecycle node.

I tested these parameters by substituting the info from our map on the resize step, and it seems to work properly even when the `static layer` plugin is removed. You're right that the static layer plugin will update the map bounds according to the `map_server`, but this method would be an alternative. I believe supporting these parameters again would allow us to not rely on the static layer plugin and even use the stack without the map server (only with obstacle and inflation layer) without the requiring the rolling window.

bpwilcox, 2019-07-10:
I was originally thinking that when the static layer is not enabled it should behave the same as if the plugin was not loaded. However, there are some consequences to having the static layer plugin loaded (even if disabled) such as having incoming maps from map_server update the size of the underlying costmap. I'm not sure I agree it should be the static_layer's sole responsibility to do that, and that seemingly wasn't the case in the ROS1 stack. 

This may serve another motivation to re-enable dynamic parameters and allow for the map parameters to be updated and change the costmap size as an option instead of using the static layer plugin. @crdelsey Thoughts?

crdelsey, 2019-07-11:
@bpwilcox Can you think of a reason someone would want to dynanically change the costmap size if they don't have the static layer plugin?

SteveMacenski, 2019-11-14:
Our discussions said to add a warning when the robot is off the costmap

closed 2019-11-22 by SteveMacenski