ros-navigation/navigation2 #904: BT Navigator Crash

crdelsey, 2019-07-01:
# Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 0.2.0
- DDS implementation:
  - FastRTPS
- Client library (if applicable):
  - 

#### Steps to reproduce issue

After navigating around for a few minutes, BT Navigator crashed. Here's the log output
```
[dwb_controller-5] [INFO] [dwb_controller]: Preempting the goal. Passing the new path to the planner.
[dwb_controller-5] [ERROR] [DWBLocalPlanner]: No valid trajectories out of 398! 
[dwb_controller-5] [ERROR] [DWBLocalPlanner]: 1.00: BaseObstacle/Trajectory Hits Obstacle.
[dwb_controller-5] [ERROR] [dwb_controller]: No valid trajectories out of 398! 
[navfn_planner-6] [WARN] [navfn_planner]: Planning algorithm failed to generate a valid path to (1.94, 0.28)
[bt_navigator-8] terminate called after throwing an instance of 'rclcpp_action::exceptions::UnknownGoalHandleError'
[bt_navigator-8]   what():  Goal handle is not know to this client.
[ERROR] [bt_navigator-8]: process has died [pid 7761, exit code -6, cmd '/home/crdelsey/src/ros_dev_workspace/navigation2_ws/install/lib/nav2_bt_navigator/bt_navigator __node:=bt_navigator __params:=/tmp/tmpcpxe17ut']
```

Looks like DWB failed and then Navfn failed. I'm not sure if that is the condition that led to the failure.

orduno, 2019-07-03:
@crdelsey Do you recall if this issue occurred when providing a new goal point in RVIZ before the current goal was completed?

bpwilcox, 2019-07-11:
Fixed with #917

closed 2019-07-11 by bpwilcox