ros-navigation/navigation2 #905: Rviz cannnot response when pressing the navigation2 goal button

MengNan-Li, 2019-07-02:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu18.04
- Installation type:
  - binaries
- Version or commit hash:
  - 0.2.0
- DDS implementation:
  -  Fast-RTPS
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
(1) start our robot driver.
(2) ros2 launch nav2_bringup nav2_bringup_launch.py autostart:=True map:=**.yaml use_sim_time:=False map_type:=occupancy
(3) ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/launch/nav2_default_view.rviz
(4) rviz set initial pose successfully
(5) When seting the navigation2 goal, the rviz cannot response and crash.
```

#### Expected behavior
Robot move the goal point set in rviz  navigation2 goal button.
#### Actual behavior
When seting the navigation2 goal, the rviz cannot response and crash.
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

mlherd, 2019-07-02:
Hi MengNan! I have a few questions. What robot do you have? Can you see the map after launching Rviz? Do you see any error messages in any of the terminals after setting the initial pose and the goal position?

testkit, 2019-07-03:
Similar issue was found in rviz2 for simulation, but this rviz2 issue might be the curlprit https://github.com/ros2/rviz/issues/417.

MengNan-Li, 2019-07-03:
@mlherd We use our own robots([YUNJI Technology](http://en.yunjichina.com.cn/)). I can the map and set the initial pose of robot.
[image]

The log messages of setting the initial pose:
[amcl-3] [WARN] [amcl]: Waiting for map....
[amcl-3] [INFO] [amcl]: initialPoseReceived
[amcl-3] [INFO] [amcl]: Setting pose (1562135076.330833): 0.575 1.877 -1.570
[amcl-3] [INFO] [amcl]: Received a 384 X 384 map @ 0.050 m/pix
[amcl-3] 
[amcl-3] [INFO] [amcl]: Setting pose (1562135077.010127): 0.575 1.877 -1.570
[amcl-3] [INFO] [amcl]: createLaserObject

MengNan-Li, 2019-07-03:
When seting the navigation2 goal, the rviz cannot response. And the terminal has no new log information.
[image]

mkhansen-intel, 2019-07-03:
Hi @MengNan-Li - I'm sorry to see you're having issues. We have seen some issues with Rviz2, but let's see if the nav2 system is running correctly in this case.
1) Is Gazebo still running? can you verify that the simulation hasn't crashed for some reason?
2) Is there any log output after AMCL initialized, from the lifecycle_manager node? 
-- You should see it 'activating' the rest of the nodes in the system, and eventually see a message that says 'The system is active'. This will tell you that all the nodes were successfully transitioned to the 'active' lifecycle state. If you don't see that, there is a possibility that something failed. 
3) Another thing to check is whether the nodes are running. When you do a 'ros2 node list' what do you get? The bt_navigator node is the one that is providing the action server for the navigation2 button. Make sure that node is running and active. You can get its state by 'ros2 service call /bt_navigator/get_state lifecycle_srvs/GetState'

MengNan-Li, 2019-07-04:
Hi @mkhansen-intel Thinks for your response.
1. I use our own real robot, not gazebo. 
2. The log information after starting the navigation launch file is as follows:
[image]
The following information is continuously refreshed until acml is initialized.
[image]
3. ros2 node list:
[image]
ros2 service call /bt_navigator/get_state lifecycle_msgs/srv/GetState:
[image]

MengNan-Li, 2019-07-09:
Think you all. When the issue(#908) is solved, the problem is gone.

closed 2019-07-09 by MengNan-Li