ros-navigation/navigation2 #907: Remove 'extra' branches

mkhansen-intel, 2019-07-03:
More of a 'heads-up' than a bug: 
 **I would like to remove all the 'extra' branches that were used for staging PRs**
 - examples: lifecycle, tb3_integration, etc.
 - The only branches that would be left would be:
 -- master (default)
 -- crystal-devel
 -- dashing-devel

Heads-up: @SteveMacenski, @mjeronimo, @crdelsey and anyone else I may have missed

mjeronimo, 2019-07-05:
I'm fine with deleting the lifecycle branch; obsolete now that its integrated.

mkhansen-intel, 2019-07-08:
I deleted lifecycle and all the branches that were either 'closed' or merged already.

remaining to-do:

- [ ] catch_references: Updated 7 days ago by @bpwilcox ; #706  Open

- [ ] robot_class_to_util: Updated 10 days ago by @bpwilcox ; #704  Open

- [ ] primatives_to_recoveries: Updated 2 months ago by @SteveMacenski ; #705  Open

- [ ] tb3_integration: Updated 3 months ago by @crdelsey -- Can we delete this now?

For those that have Open PRs, we can leave them for now until the PRs are merged. In the future, let's not create branches in the repo for PRs, let's create branches in our forks then create a PR from there to master.

crdelsey, 2019-07-08:
tb3_integration is now deleted

SteveMacenski, 2019-07-11:
Ok, sorry about that. I think I was the most egregious user or branches on origin. The workflow I usually follow is branches in repos I have write access to or maintain, forks when either of those criteria are false.

Won't (intentionally) happen again.

SteveMacenski, 2019-07-16:
Last 2 are on me, both need some additional comments from reviewers:
- I need the "ok" to rename primitives to recoveries from more than 1 person
- I need the "ok" to kill all recovery use of positioning information in favor of odometry for reasons listed in the PR. All things creating control signals should utility odometry

SteveMacenski, 2019-07-19:
done with them

closed 2019-07-19 by SteveMacenski