ros-navigation/navigation2 #908: Rviz crash when load our own map.

MengNan-Li, 2019-07-03:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu18.04
- Installation type:
  - binaries
- Version or commit hash:
  - 0.2.0
- DDS implementation:
  -Fast-RTPS, OpenSplice
- Client library (if applicable):
  - rclcpp

#### Steps to reproduce issue

```
Use turtlebot3_world map to display properly in Rviz: 
ros2 launch nav2_bringup nav2_bringup_launch.py autostart:=True map:=/home/ros/yunji_ros2/src/navigation2/nav2_bringup/launch/turtlebot3_world.yaml use_sim_time:=False map_type:=occupancy

But, Using our own map doesn't display properly:
ros2 launch nav2_bringup nav2_bringup_launch.py autostart:=True map:=/home/ros/yunji_ros2/src/navigation2/nav2_bringup/launch/11/map1.yaml use_sim_time:=False map_type:=occupancy

ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/launch/nav2_default_view.rviz

```

#### Expected behavior
Showing the map in Rviz
#### Actual behavior
ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/launch/nav2_default_view.rviz
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 3 (GLSL 1.3)
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: Creating 1 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 1 swatches_
[INFO] [rviz2]: Creating 2 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 2 swatches_
[INFO] [rviz2]: Creating 4 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 4 swatches_
[INFO] [rviz2]: Creating 8 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
[ERROR] [rviz2]: Failed to create 8 swatches_
[ERROR] [rviz2]: InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)
terminate called after throwing an instance of 'Ogre::InvalidParametersException'
  what():  InvalidParametersException: Stream size does not match calculated image size in Image::loadRawData at /tmp/binarydeb/ros-dashing-rviz-ogre-vendor-6.1.1/obj-x86_64-linux-gnu/ogre-master-ca665a6-prefix/src/ogre-master-ca665a6/OgreMain/src/OgreImage.cpp (line 283)

#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

MengNan-Li, 2019-07-03:
The map file:
git clone https://github.com/MengNan-Li/map.git

mkhansen-intel, 2019-07-03:
@MengNan-Li - I've seen these error messages before with Rviz2, I think there is an Rviz2 bug here. Usually those messages seem to be harmless, but in your case is rviz2 actually crashing? If so, if you restart rviz2 does it crash again? Also, is the nav2 bringup window showing any error messages? If things are working correctly, you should see the lifecycle_manager bring up map_server and amcl to the active state. It should then be printing messages that it's waiting for an initial pose.

MengNan-Li, 2019-07-04:
@mkhansen-intel - I don't know if it is a map format problem. Because Rviz2 can display map normally when i load navigation2/nav2_bringup/launch/turtlebot3_world.yaml. You can test this using the [map](https://github.com/MengNan-Li/map.git) i provided.
 
```
1. ros2 launch nav2_bringup nav2_bringup_launch.py map:=/home/ros/yunji_ros2/src/navigation2/nav2_bringup/launch/map1.yaml
2. ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/launch/nav2_default_view.rviz

```

rotu, 2019-07-05:
I think this is related to the map image dimensions. I have isolated a similar crash down to only changing the map size and scale. https://github.com/ros2/rviz/issues/424

MengNan-Li, 2019-07-08:
@rotu Think you update the issue. [Martin-Idel's branch](https://github.com/Martin-Idel/rviz/tree/bugfix/map_display)  can isolate the crash down.

closed 2019-07-08 by MengNan-Li

mkhansen-intel, 2019-07-08:
Thanks for the help @rotu!