ros-navigation/navigation2 #909: MotionPrimitiveNode registers mulitple subnodes with same name

rotu, 2019-07-03:
When spinning up MotionPrimitiveNode, 
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18
- Installation type:
  - source
- Version or commit hash:
  - d719479d994a04150eb4c3568ee7d6f2fbb8c737
- DDS implementation:
  - OpenSplice
- Client library (if applicable):
  - N/A

#### Steps to reproduce issue
```
ros2 run nav2_motion_primitives motion_primitives_node
```

#### Expected behavior
Starts up with no problem

#### Actual behavior
Emits a warning:
```
[INFO] [motion_primitives]: Configuring Spin
[INFO] [motion_primitives]: Robot: Creating
[WARN] [rcl.logging_rosout]: Publisher already registered for provided node name. If this is due to multiple nodes with the same name then all logs for that logger name will go out over the existing publisher. As soon as any node with that name is destructed it will unregister the publisher, preventing any further logs for that name from being published on the rosout topic.
[INFO] [motion_primitives]: Configuring BackUp
[INFO] [motion_primitives]: Robot: Creating
```
The duplicate nodes can be seen in `ros2 node list`:
```
/motion_primitives
/motion_primitive_GetRobotPose_client
/motion_primitive_GetRobotPose_client
```

#### Additional information
This would be resolved by fixing #816

mkhansen-intel, 2019-07-03:
So we agree this is an undesired node, but can you tell me if this is causing any errors? I'm trying to get a feel for how high priority this is right now.

rotu, 2019-07-05:
@mkhansen-intel as far as I can tell the Spin and BackUp nodes still work properly, and the only problem this causes is the warning in the log.

SteveMacenski, 2019-08-08:
@rotu by no means should you not merge that PR but my TF PR removes the `GetRobotPose` server, services, and messages. If that was the issue its gone. 

https://github.com/ros-planning/navigation2/pull/993

rotu, 2019-08-08:
@SteveMacenski This issue as described is closed. But my PR also fixes `ServiceClient::ServiceClient(const std::string & service_name, const std::string & parent_name)` so it shouldn't result in such collisions in general. Also, fixes a regression due to upstream changes in rcl.

closed 2019-08-08 by rotu