ros-navigation/navigation2 #911: Robot continues moving when pressing "cancel" button during navigation (happens only sometimes)

AndreasAZiegler, 2019-07-08:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.10
- Installation type:
  - binaries
- Version or commit hash:
  - dashing-devel da5360af1e69cabe394c1eb15191b1e0fb630d16
- DDS implementation:
  - Fast-RTPS and RTI Connext

#### Steps to reproduce issue
* Follow steps in https://github.com/ros-planning/navigation2/tree/dashing-devel/nav2_bringup
* Press "cancel" button during navigation

#### Expected behavior
Robot stops moving.

#### Actual behavior
Sometimes robots keep moving and starts toddling around after colliding with an obstacle

crdelsey, 2019-07-08:
@AndreasAZiegler Can you include the console output? This sounds like something in the nav stack has crashed, preventing us from sending the final 0 velocity command.

orduno, 2019-07-08:
I followed the steps above and am not seeing the issue.

@AndreasAZiegler How often do you see this behavior?

AndreasAZiegler, 2019-07-09:
@orduno Yes, my console output is similar.

```
[bt_navigator-8] [INFO] [bt_navigator]: Begin navigating from current location to (2.09, -1.49)
[dwb_controller-5] [INFO] [dwb_controller]: Received a goal, begin following path
[dwb_controller-5] [INFO] [dwb_controller]: Preempting the goal. Passing the new path to the planner.
[dwb_controller-5] [INFO] [dwb_controller]: Preempting the goal. Passing the new path to the planner.
[dwb_controller-5] [INFO] [dwb_controller]: Preempting the goal. Passing the new path to the planner.
[dwb_controller-5] [INFO] [dwb_controller]: Preempting the goal. Passing the new path to the planner.
[bt_navigator-8] [INFO] [bt_navigator]: Navigation canceled
[dwb_controller-5] [INFO] [dwb_controller]: Goal was canceled. Cancelling and stopping.
[dwb_controller-5] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[dwb_controller-5]   what():  goal_handle attempted invalid transition from state EXECUTING with event CANCELED, at /tmp/binarydeb/ros-dashing-rcl-action-0.7.5/src/rcl_action/goal_handle.c:94
[ERROR] [dwb_controller-5]: process has died [pid 8479, exit code -6, cmd '/home/andreasziegler/ros/ros2_turtlebot3_ws/install/dwb_controller/lib/dwb_controller/dwb_controller __params:=/tmp/tmpqs_jcrgi'].
```

orduno, 2019-07-09:
@AndreasAZiegler I've addressed the bug on #918, could you please try it and let me know if that fixes the issue on your side?

AndreasAZiegler, 2019-07-10:
@orduno I tried it a couple of times and it seems to work. Thanks.

orduno, 2019-07-11:
Closing the issue as #918 was merged.

closed 2019-07-11 by orduno