ros-navigation/navigation2 #913: Introducing Reinforcement Learning to ROS2 Navigation Stack

mhpanah, 2019-07-08:
Currently, ROS2 Navigation stack uses Dynamic Windo Approach (DWA) as a core obstacle avoidance for the local planner. Though this planner has been widely used over the years it has several shortcomings such as having many parameters that need to be tuned manually, sensitive to critic parameters, and lack of intelligence to recover from failures.

To overcome these drawbacks, we plan to bring AI to ROS 2 Navigation Stack gradually to replace different components of the Navigation module.  First, we introduced the Random Crawl task to navigate around without hitting any obstacles with pseudo-random actions. Now we plan to extend upon this task and investigate different actions to be able to replace the current local planner with AI. This task will be explored and implemented in the following stages:

- [ ] Train a differential drive robot with LIDAR in a static environment (E Turtle Release)

- [ ] Train a differential drive robot with LIDAR in a dynamic environment (E Turtle Release)

- [ ] Investigate different accelerators such as Moviduis or others, run the planner and compare the performance

- [ ] Integrate Camera and Object Analytics in addition to LIDAR

mhpanah, 2020-01-13:
This work is currently halted.  I will close for now.

closed 2020-01-13 by mhpanah

SteveMacenski, 2020-01-14:
@mhpanah what should we do with the DQN work inside of `nav2_experimental`? I'd vote to throw it into another branch `nav2_experimental` so we stop releasing it if its on hold & doesn't appear to users as an element of the core metapackage.