ros-navigation/navigation2 #915: Gazebo crashes when using RTI Connext

AndreasAZiegler, 2019-07-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.10
- Installation type:
  - binaries
- Version or commit hash:
  - dashing-devel da5360a
- DDS implementation:
  - RTI Connext

#### Steps to reproduce issue
* Follow steps in https://github.com/ros-planning/navigation2/tree/dashing-devel/nav2_bringup
* When I try to start Gazebo, it crashes

#### Expected behavior
Gazebo starts normally

#### Actual behavior
Gazebo crashes with the following console output

```
andreasziegler@AndreasZiegler:~/ros/ros2_turtlebot3_ws$ ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py
[INFO] [launch]: All log files can be found below /home/andreasziegler/.ros/log/2019-07-09-08-40-46-230551-AndreasZiegler-6497
[INFO] [launch]: Default logging verbosity is set to INFO
urdf_file_name : turtlebot3_burger.urdf
[INFO] [gazebo-1]: process started with pid [6517]
[INFO] [ros2-2]: process started with pid [6518]
[INFO] [robot_state_publisher-3]: process started with pid [6519]
[robot_state_publisher-3] Initialize urdf model from file: /home/andreasziegler/ros/ros2_turtlebot3_ws/install/turtlebot3_description/share/turtlebot3_description/urdf/turtlebot3_burger.urdf
[robot_state_publisher-3] Parsing robot urdf xml string.
[robot_state_publisher-3] Link base_link had 5 children
[robot_state_publisher-3] Link caster_back_link had 0 children
[robot_state_publisher-3] Link imu_link had 0 children
[robot_state_publisher-3] Link base_scan had 0 children
[robot_state_publisher-3] Link wheel_left_link had 0 children
[robot_state_publisher-3] Link wheel_right_link had 0 children
[robot_state_publisher-3] got segment base_footprint
[robot_state_publisher-3] got segment base_link
[robot_state_publisher-3] got segment base_scan
[robot_state_publisher-3] got segment caster_back_link
[robot_state_publisher-3] got segment imu_link
[robot_state_publisher-3] got segment wheel_left_link
[robot_state_publisher-3] got segment wheel_right_link
[robot_state_publisher-3] Adding fixed segment from base_footprint to base_link
[robot_state_publisher-3] Adding fixed segment from base_link to caster_back_link
[robot_state_publisher-3] Adding fixed segment from base_link to imu_link
[robot_state_publisher-3] Adding fixed segment from base_link to base_scan
[robot_state_publisher-3] Adding moving segment from base_link to wheel_left_link
[robot_state_publisher-3] Adding moving segment from base_link to wheel_right_link
[gazebo-1] Gazebo multi-robot simulator, version 9.9.0
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[gazebo-1] Gazebo multi-robot simulator, version 9.9.0
[gazebo-1] Copyright (C) 2012 Open Source Robotics Foundation.
[gazebo-1] Released under the Apache 2 License.
[gazebo-1] http://gazebosim.org
[gazebo-1] 
[gazebo-1] terminate called after throwing an instance of 'rclcpp::exceptions::RCLError'
[gazebo-1]   what():  could not create publisher: failed to create topic, at /tmp/binarydeb/ros-dashing-rmw-connext-cpp-0.7.3/src/rmw_publisher.cpp:199, at /tmp/binarydeb/ros-dashing-rcl-0.7.5/src/rcl/publisher.c:171
[ERROR] [gazebo-1]: process has died [pid 6517, exit code 255, cmd 'gazebo --verbose /home/andreasziegler/ros/ros2_turtlebot3_ws/install/turtlebot3_gazebo/share/turtlebot3_gazebo/worlds/turtlebot3_worlds/burger.model -s libgazebo_ros_init.so'].
[ros2-2] Node not found
[ERROR] [ros2-2]: process has died [pid 6518, exit code 1, cmd 'ros2 param set /gazebo use_sim_time true'].
```

crdelsey, 2019-07-11:
I've managed to reproduce the problem and am investigating it.

SteveMacenski, 2019-10-15:
Closing from inactivity + several RTI updates since July.

closed 2019-10-15 by SteveMacenski

crdelsey, 2019-10-16:
Sorry. I hadn't had a chance to get back to this yet. I did retest, and I am still able to reproduce the problem, however, I can reproduce it without any navigation2 in the environment. This does seem to be a gazebo or rmw_connext problem.

SteveMacenski, 2019-10-16:
I'll stick with "more of a gazebo_ros ticket" then