ros-navigation/navigation2 #916: rviz does not start when using RTI Connext

AndreasAZiegler, 2019-07-09:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.10
- Installation type:
  - binaries
- Version or commit hash:
  - dashing-devel da5360a
- DDS implementation:
  - RTI Connext

#### Steps to reproduce issue
* Follow steps in https://github.com/ros-planning/navigation2/tree/dashing-devel/nav2_bringup
* When I try to start rviz it does not.

#### Expected behavior
Rviz starts

#### Actual behavior
I get the following console output

```
andreasziegler@AndreasZiegler:~/ros/ros2_turtlebot3_ws$ ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/launch/nav2_default_view.rviz
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-andreasziegler'
[INFO] [rviz2]: Stereo is NOT SUPPORTED
[INFO] [rviz2]: OpenGl version: 4.6 (GLSL 4.6)
[INFO] [rviz2]: Stereo is NOT SUPPORTED
```
Stack trace is attached.
[my_log_file.txt](https://github.com/ros-planning/navigation2/files/3372113/my_log_file.txt)

crdelsey, 2019-07-09:
@AndreasAZiegler We have briefly tried RTI Connext and found there are limits in that DDS implementation that prevent us from creating enough nodes to run the full navigation stack. I believe that is what is happening in this issue and #915 

I will try to reproduce this with a simplified example and file this upstream.

A few questions to make sure I have the same configuration:
1.  What type of license are you using? Is it the 30 day trial? 
2. Have you made any configuration changes to connext? 
3. Are you running with the `ROS_DOMAIN_ID` environment variable set to avoid interference with others on your network?

AndreasAZiegler, 2019-07-10:
1. We have a commercial license from rti but I used the normal debian package (and copied our license file)
2. No
3. No, I did not set ```ROS_DOMAIN_ID``` as I was not aware of this. How can I use this? As we have quite a lot of other devices using plain rit Connext DDS it could actually be worth a try.

crdelsey, 2019-07-10:
Here's the info on ROS_DOMAIN_ID https://index.ros.org/doc/ros2/Contributing/ROS-2-On-boarding-Guide/#choose-a-dds-domain-id

We ran into problems with that. Each developer would interfere with the others on the network producing strange results.

SteveMacenski, 2019-08-24:
@AndreasAZiegler @crdelsey this ticket doesn't seem to directly relate to this project, can we close or redirect the issue to the appropriate projects (Rviz)?

AndreasAZiegler, 2019-08-24:
I'm fine if someone redirects this issue to the appropriate project. In case someone following this issue comes up with a solution I would be glad if he/she would write about the solution here (as well).

mkhansen-intel, 2019-08-26:
@AndreasAZiegler - I think this is a duplicate of this issue:
https://github.com/ros2/rviz/issues/437