ros-navigation/navigation2 #920: The robot continuous trying to reach to the previous goal pose after setting a new goal pose

mlherd, 2019-07-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - Built from source
- Version or commit hash:
  - master

#### Steps to reproduce issue

```
- Launch Gazebo
- Launch Turtlebot3 bringup
- Launch Navigation2 bringup
- Launch Rviz
- Set an initial pose
- Set a goal pose
- Set another goal pose before the robot completes the previous one  
```

#### Expected behavior

- I can think of two possible expected behaviors.

1- Robot reaches the first goal pose and than starts moving towards the next one.
2- Robot immediately stops if there is a new goal pose and start following the new global path that goes to the new goal pose. 

#### Actual behavior

It calculates a new global path after selecting a new goal pose, but the robot keeps moving towards the previously selected goal pose.

orduno, 2019-07-10:
@mlherd How are you setting the goal pose on RVIZ? Using the `Navigation2 Goal` or `2D Nav Goal`? I don't see the issue with `Navigation2 Goal`. How often do you see this problem? Could you post the console output?

Sometimes the planner will actually compute the path towards the new goal, however, if the controller is close to the previous goal and succeeds before the preemption is communicated, navigation will succeed on the previous goal.

mlherd, 2019-07-10:
It seems like when I had this issue I wasn't using the latest version of the stack. I updated everything and I tested it again. This problem does not exist anymore.

crdelsey, 2019-07-30:
I'm seeing this behavior still with the 0.2.3 release candidate.

crdelsey, 2019-07-31:
Looks like it only happens on the first nav goal. After it gets to the first goal and you give a second goal, the preemption works as expected.

orduno, 2019-10-30:
The issue has been coming and going. Let's figure out on #1301 what's the problem on the current master branch.

closed 2019-10-30 by orduno