ros-navigation/navigation2 #921: Local costmap not displayed (Could not transform from [odom] to [map])

AndreasAZiegler, 2019-07-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.10
- Installation type:
  - binaries
- Version or commit hash:
  - dashing-devel da5360af1e69cabe394c1eb15191b1e0fb630d16
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
* Follow steps in https://github.com/ros-planning/navigation2/tree/dashing-devel/nav2_bringup

#### Expected behavior
Rviz displays the local map if enabled

#### Actual behavior
Rviz can't display the local map. I says "Could not transform from [odom] to [map]" even though rviz shows the tf tree where map->odom is contained

[image]

crdelsey, 2019-07-11:
There are problems with Rviz displaying data at the moment. I believe it doesn't use message filters for all messages yet. The result is that Rviz tries to transform incoming data but the transformation hasn't arrived yet, so it just displays an error message instead.

I just checked and I see this is a known problem with Rviz. It is mentioned on their front page among "other features" to be ported. https://github.com/ros2/rviz

crdelsey, 2019-07-11:
Looks like there is a PR for Rviz already. ros2/rviz#375

closed 2019-07-11 by crdelsey

ChristofGroschke, 2020-06-09:
Hey, has this issue been solved ? The PR is still on in review. Thanks.

ChristofGroschke, 2020-06-09:
Quick question. Is this already fixed. I think i have a similar problem with turtlebot3, kinect and cartographer. Cartographer is running in localization mode. The global costmap is showing up but the for the local costmap rviz reports no transform from /odom to /map. The transform is valid, and is showing of under tf tree. If it's is fixed does someone now if this is already upstream in the Ubuntu 18.04 repositories ? Or do i have to use the devel branch ? Thanks. Sorry for opening this up.

SteveMacenski, 2020-06-09:
Ask on ROS Answers