ros-navigation/navigation2 #922: Rviz does not display laser scans

AndreasAZiegler, 2019-07-10:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - binaries
- Version or commit hash:
  - dashing-devel da5360af1e69cabe394c1eb15191b1e0fb630d16
- DDS implementation:
  - Fast-RTPS

#### Steps to reproduce issue
* Follow steps in https://github.com/ros-planning/navigation2/tree/dashing-devel/nav2_bringup

#### Expected behavior
Rviz displays the laser scans if enabled

#### Actual behavior
Rviz can't display the laser scans. The following is printed in the console output:
```
[ERROR] [rviz2]: Lookup would require extrapolation into the future.  Requested time 1004.61100 but the latest data is at time 1004.56400, when looking up transform from frame [base_scan] to frame [map]
```

crdelsey, 2019-07-10:
This is a problem with RViz. I expect it hasn't enabled message filters yet.

SteveMacenski, 2019-07-10:
Thanks for filing the ticket, but this belongs in the Rviz project issue tracker @AndreasAZiegler. Please file a ticket so the right maintainers can take a look!

closed 2019-07-10 by SteveMacenski

crdelsey, 2019-07-11:
There is a PR outstanding for Rviz that re-enables message filters ros2/rviz#375 I believe that will fix the problem once it is integrated.