ros-navigation/navigation2 #925: Rviz2 will crash if robot hit the post in navigate (in simulate environment), have few rate navigate will not auto cancel

rafaelhuang-intel, 2019-07-11:
## Bug report

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
Add gazebo and turtlebot module to source 

1.gazebo --verbose -s libgazebo_ros_init.so /home/intel/robot_devkit/rdk_ws/Turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_worlds/waffle.model

2.ros2 launch turtlebot3_bringup turtlebot3_state_publisher.launch.py use_sim_time:=True

3.ros2 launch nav2_bringup nav2_bringup_launch.py use_sim_time:=True autostart:=True map:=/home/intel/robot_devkit/rdk_ws/Turtlebot3_ws/src/navigation2/navigation2/nav2_bringup/launch/turtlebot3_world.yaml

4. ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/launch/nav2_default_view.rviz

5. press start button on Rviz
6. set initial pose
7. set goal from A to B
8. Check robot status in Rviz & Gazebo
```

#### Expected behavior
Robot can navigate from A to B and avoid post.
#### Actual behavior
if robot hit the post in navigate, have few rate navigate will not auto cancel, in this case Rviz2 will crash.
#### Additional information
[error_message_log.txt](https://github.com/ros-planning/navigation2/files/3380149/error_message_log.txt)

----
## Feature request

#### Feature description

#### Implementation considerations

crdelsey, 2019-07-11:
I believe #917 and #918 will fix this problem. They should both be integrated today, I believe.

orduno, 2019-07-11:
@rafaelhuang-intel we've merged #917 and #918, could you check if the issue is solved?

rafaelhuang-intel, 2019-07-12:
@orduno I see #918 already merged to master branch, but currently my test environment is dashing-devel +ros2 binary 0614, so I will try it later.

attached new log, I meet this issue again that rviz will frozen long times before Rviz crash.

[error_message_log_2.txt](https://github.com/ros-planning/navigation2/files/3384783/error_message_log_2.txt)

mlherd, 2019-07-12:
@rafaelhuang-intel I noticed the issue that rviz gets frozen today again. After rviz staying frozen for a while it just crashes. I don't know the exact causes yet, but it happens after every a few successful launches. #919 fixes the crashing issue for a few different scenarios such as if gazebo is not launched and the initial pose is not set, but it seems like there are some other issues causing the crash as well. I had a chance to investigate issue a little bit. It seems like the crash happens during another action call in nav2 rviz plugin. I will keep working on this issue.

mlherd, 2019-07-24:
#937 addresses this issue. Rviz plugin should no longer cause RVIZ to get frozen or crashed.

closed 2019-07-24 by mlherd

rafaelhuang-intel, 2019-07-25:
@mlherd currently we use ROS2 dashing 0614 (binary) + Navigation2 / dashing-devel version, so could you help to merge this PR to dashing-devel branch?

mlherd, 2019-07-25:
@rafaelhuang-intel Sure! Actually there are many ways to do it; I guess this PR will be merged to master tomorrow.

I guess one way to do is this;

1- Go to the Navigation 2 repository in my profile
`https://github.com/mlherd/navigation2/tree/rviz-plugin`

2- clone the rviz-plugin branch to some directory in your computer
`git clone --branch rviz-plugin https://github.com/mlherd/navigation2.git`

2-delete the nav2_rviz_plugin package in your current navigation2 workspace

3- move the new nav2_rviz_plugin package to the navigation 2 workspace

4- build the navigation 2 stack
`colcon build --symlink-install`

Let me know if you have any questions.

rafaelhuang-intel, 2019-07-25:
@mlherd 
suggestion you can merge this PR to dashing-devel, like (https://github.com/ros-planning/navigation2/pull/918) , because above step not easy to deliver more test platform.

mlherd, 2019-07-25:
@rafaelhuang-intel I thought you wanted to have the changes now. That is why I suggested you this way. I can't merge the PR to the master or dashing-devel branches without approval. I think the newest features and changes will always be merged to master first.

crdelsey, 2019-07-25:
@rafaelhuang-intel We're planning an update of the dashing-devel branch next week.

rafaelhuang-intel, 2019-07-26:
@crdelsey 

Thanks for your update, I will double check this issue on ROS2 dashing 0614 (binary) + Navigation2 / dashing-devel version while devel branch update at next week.