ros-navigation/navigation2 #926: if set goal from A to B, navigate will auto stop and robot not arrive expect point

rafaelhuang-intel, 2019-07-11:
## Bug report

I meet problem that if set goal from A to B, navigate will auto stop and robot not arrive expect point, but I can set goal again and robot workable except hit the post or wall, this issue meet rate ~30% ~ 40%, so do you meet same problem?

**Required Info:**

- Operating System:
  - Ubuntu 18.04
- Installation type:
  - source
- Version or commit hash:
  - dashing-devel 43f8e726b9bab2eda5767203d5b30ab64c1ad69d
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue

```
Add gazebo and turtlebot module to source 

1.gazebo --verbose -s libgazebo_ros_init.so /home/intel/robot_devkit/rdk_ws/Turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/worlds/turtlebot3_worlds/waffle.model

2.ros2 launch turtlebot3_bringup turtlebot3_state_publisher.launch.py use_sim_time:=True

3.ros2 launch nav2_bringup nav2_bringup_launch.py use_sim_time:=True autostart:=True map:=/home/intel/robot_devkit/rdk_ws/Turtlebot3_ws/src/navigation2/navigation2/nav2_bringup/launch/turtlebot3_world.yaml

4. ros2 run rviz2 rviz2 -d $(ros2 pkg prefix nav2_bringup)/share/nav2_bringup/launch/nav2_default_view.rviz

5. press start button on Rviz
6. set initial pose
7. set goal from A to B
8. Check robot status in Rviz & Gazebo
```

#### Expected behavior
 if set goal from A to B, robot arrive expect point
#### Actual behavior
 if set goal from A to B, navigate will auto stop and robot not arrive expect point
#### Additional information

----
## Feature request

#### Feature description

#### Implementation considerations

crdelsey, 2019-07-11:
> I meet problem that if set goal from A to B, navigate will auto stop

Does the `error_message_log.txt` that you attached to #925 also apply to this problem? If so, I think #917 will fix it.

mlherd, 2019-07-12:
I had the same issue today. Also Rviz still crashes time to time. I will keep investigating this issue.

mlherd, 2019-07-16:
@rafaelhuang-intel 
I have run so many tests, but still I am not able to regenerate this issue. Do you still have this problem?

rafaelhuang-intel, 2019-07-16:
@mlherd 

I have tried Ros2 core 0614 + Navigation2 / Dashing-devel (016df2945cf9f9960c0e0416746ef504c8bcf8af), this issue still can reproduced.

attached log and screen capture:

[image]
[bug_926.txt](https://github.com/ros-planning/navigation2/files/3395772/bug_926.txt)

mlherd, 2019-07-16:
@rafaelhuang-intel 
Thanks for the video and the log file. I will investigate this issue.

crdelsey, 2019-07-16:
#936 is a possible workaround for this, I think. Looks like the robot is just getting too close to obstacles in the video

SteveMacenski, 2019-07-16:
An aside but something to bring up: collectively, a lot of time has been spent on DWB and it doesnt seem that anyone is happy with it. I propose we adapt it in the short term but talk about what it might look like to remove it entirely from support. MPC is a very popular technique for a reason and should be somewhat simple to generalize across different robot base types. 

The DRL work is interesting to @mhpanah  and me but I recognize that even if I make it work to a degree I am happy with, its research code that wouldn't find its way to prime time for quite awhile until the RL community starts leveling out. 

Thoughts?

crdelsey, 2019-07-17:
I fully agree. I wanted to get the DWA/B algorithm approach to match the quality achieved in ROS 1 before spending time to find a better algorithm. There's no reason it shouldn't work as well or better than DWA/base_local_planner in ROS 1. 

If we can get a local planner interface defined like you mentioned in some other issue, that would be a good point to get all the ROS2 changes to DWB pushed back to @DLu's repo. Then navigation2 could just provide the core that DWB, TEB, MPC and others use.

bpwilcox, 2019-07-17:
Highly in favor of such a local planner interface!

SteveMacenski, 2019-07-17:
Ticket here https://github.com/ros-planning/navigation2/issues/943

Easier to make these changes now before trying to port multiple local or global planners. My recommendation would be to stop porting TEB until we have defined something *unless someone actually needs it*

bpwilcox, 2019-07-17:
> Ticket here #943
> 
> Easier to make these changes now before trying to port multiple local or global planners. My recommendation would be to stop porting TEB until we have defined something _unless someone actually needs it_

@yathartha3 has been looking into the port of TEB. Perhaps he can also look into this.

orduno, 2019-07-17:
I suggest we take some time to discuss the design and clarify the interface, integration components (plugins, adaptors, ROS wrapper), rethink role of a local planner (pure path following & collision avoidance), and interaction with global planning (MPC can do both tasks) and set the stage for further work on enabling multi-robot (traffic control, etc.).

yathartha3, 2019-08-19:
I submitted a PR #1045 regarding issue #943 .
This has the local planner interface and follow path action interface. It works with my turtlebot3 in simulation, however some more testing might be necessary.

mlherd, 2019-08-24:
I have been testing Navigation2 (Dashing) with a real robot (Turtlebot 3). I almost never see the robot hitting obstacles or getting too close to them.

@rafaelhuang-intel Do you still have this problem?

rafaelhuang-intel, 2019-08-26:
@mlherd 

I tested Navigation2 (Dashing) with a real robot (Turtlebot 3) & simulate by e-manual (http://emanual.robotis.com/docs/en/platform/turtlebot3/ros2/#ros2),  I almost never see this issue, so I will close it.

BTW, I see navigation 2 launch step not match dashing  release yet(https://github.com/ros-planning/navigation2/tree/master/nav2_bringup), so do you have any plan to update this one?

closed 2019-08-26 by rafaelhuang-intel