ros-navigation/navigation2 #929: The Office Marathon 2

mhpanah, 2019-07-12:
**Objective** 
Characterizing and investigating the reliability and performance of ROS 2 Navigation Stack

**Description**

The Office Maraton [1] was used on ROS 1 Navigation stack in an indoor office environment. For Navigation 2, we also need to perform similar kind of experiments to obtain better insights on the reliability, performance, and practicality of our nav stack in a real-world environment.

The following tasks are needed before conducting the Marathon Experiment:

- [ ]  Enabling Voxel Grid
- [ ]  Enabling RealSense
- [ ]  Improving Recovery Behaviors

crdelsey, 2019-07-15:
Are there other metrics we'd prefer to use to quantify our reliability?

SteveMacenski, 2019-07-15:
Checking that the local and global planners, costmaps are always making rate processing 3+ sensors would be good

SteveMacenski, 2019-07-16:
Also, let me know if you have a time scheduled for this, I can marathon in with you remotely or find my way up to Portland

mhpanah, 2019-07-16:
@SteveMacenski sounds great!  We haven't scheduled anything for this task yet.  We can talk about it more in our next WG meeting and from there schedule planning for this task.

SteveMacenski, 2020-01-14:
Anyone have interest in writing a paper for this for IROS 2020? Its a little tight of a schedule but possible. I have lasers, cameras, and a few robot bases, but no substantially large space I can use. Maybe can coerce a co-working space to let me camp out for a week or another open-source friendly place

fmrico, 2020-01-14:
Hi @SteveMacenski , I am interested in writing the paper for IROS2020, if you think it is fine, without obligation for your part. We can conduct the experiments in my lab, using kobuki or even Tiago. [We got it navigating](https://www.youtube.com/watch?v=JvGnojXBjTU) with navigation2 last week. We could extend the scenario even to the near corridors.

If you agree, you can email me to fmrico@gmail.com to coordinate.

crdelsey, 2020-01-14:
I'm likely going to have limited time to work on this soon, so I don't want to commit to this. But this is a high priority for nav 2, IMHO. If there is anything I can do to support this - new features, bugs to look into, etc, let me know.

SteveMacenski, 2020-01-15:
@fmrico sounds good! Lets chat about it at the weekly meeting tomorrow. We're moving it to 2pm because of the TSC meeting at 4pm PST.

Given the deadline is relatively soon, and to do something similar to the Office Marathon, there's going to be a pretty substantial hardware burden (and tickets filed to make it work) we should get started on this ASAP. The deadline is March 1 for regular paper submissions. That gives us about 6 weeks. 

@crdelsey I'd love it if you and the other academically minded Intel folks were to be involved. It would feel weird to not have you guys as coauthors on the paper. 

PDF of the Marathon in ROS1: http://ros.fei.edu.br/roswiki/attachments/Papers(2f)ICRA2010_Marder(2d)Eppstein/icra2010_marder-eppstein.pdf

I'd like to identify a few pieces of technology that we can use to aid in differentiators between the Marathon done back in 2010 and now. Particularly, I've ported STVL into ROS2 and that works with the stack, I think that's one example of a compelling change in algorithms. Obviously the behavior tree navigation is a big topic to cover as a differentiator. Fundamentally, not much else has really changed. Personally, I would have liked to have replaced NavFn with something that computes paths based on kinematic knowledge of the robot, removing one of the most glaring failure conditions of ROS1 navigation, but I don't think that's in the cards in the next 6 weeks. DWB is still essentially DWA that's more configurable.

SteveMacenski, 2020-03-02:
[IROS_2020___Navigation2(9).pdf](https://github.com/ros-planning/navigation2/files/4272801/IROS_2020___Navigation2.9.pdf)

Done, submitted for IROS 2020, up on Archix soon.

closed 2020-03-02 by SteveMacenski

SteveMacenski, 2020-03-03:
http://arxiv.org/abs/2003.00368

mhpanah, 2020-03-25:
@SteveMacenski Hey Steve, It's great that this work got submitted to IROS.  Let me know how the revision goes, I can help out with the BT and Recovery behavior sections.