ros-navigation/navigation2 #931: Port base local planner to ROS 2 to support migrating TEB planner

crdelsey, 2019-07-15:
Base local planner is used as a library by other planners such as TEB. We need to port it to ROS 2 to support migrating these packages to ROS 2.

yathartha3, 2019-07-18:
Ported base_local_planner is now compiling.
However, will continue working on it further after we figure out the interface.
Discussion for interface are in #943, and in comments of #926.

yathartha3, 2019-09-21:
I have a ported base_local_planner.
TEB has already been ported to ROS2, but not a plugin as of today.
We now have global and local planner interfaces in nav2_core. We can add base_local_planner functionalities if needed, into some utility folder for the planners. Closing this issue for now, reopen if needed in future.

mkhansen-intel, 2019-09-24:
I would like to have an issue to refactor TEB to use the nav2_core interface.

yathartha3, 2019-09-24:
@mkhansen-intel Reopen, and changed issue name. I pinged @vinnamkim  informing about the nav2_core interface.
There is an issue in the TEB repo: https://github.com/rst-tu-dortmund/teb_local_planner/issues/154

SteveMacenski, 2019-09-24:
I can also commit to helping (next week or after) 
Is there a timetable based on a need for this?

yathartha3, 2019-09-25:
@SteveMacenski 
> I can also commit to helping (next week or after)

👍 

> Is there a timetable based on a need for this?

Target is before ROSCon, so that we get to say we have two local planners using plugins.
Moreover, it is for parity with ROS1 (https://github.com/ros-planning/navigation2/issues/592).

SteveMacenski, 2019-09-25:
I guess that's a slippery slope to say "parity" when referring to plugins outside the main repo, how many other popular planning plugins do we need to port to call it at parity? 

With that said after reviewing the planner's entry class (https://github.com/rst-tu-dortmund/teb_local_planner/blob/dashing-devel/teb_local_planner/src/teb_controller.cpp) it shouldn't take anyone more than 3 hours to port that to `nav2_core`. Before October 30th is a pretty low bar I think we can manage to hit :wink:

SteveMacenski, 2019-09-25:
Side note: I haven't really worked much with the open source local planners - we've spun our own in most places I've been. What's the major bullet points on the reasons to use each DWB and TEB? I hear good things about TEB second hand from a number of folks. In my limited experiences with it, I never got it to behave in a way I was comfortable with, but that's a pretty weak excuse.

Is there (potentially) value in bringing TEB into the nav stack directly and/or removing DWB? If anyone has some insight on tangible pros/cons of each I'd love to have that documented for future readers and my education.

yathartha3, 2019-09-25:
> I guess that's a slippery slope to say "parity" when referring to plugins outside the main repo, how many other popular planning plugins do we need to port to call it at parity?

Just this one for right now since we put it in the roadmap. Personally, I think there are better ones out there that I would like to implement.

> we've spun our own in most places I've been.

I would really like to hear more about it.

> Is there (potentially) value in bringing TEB into the nav stack directly and/or removing DWB?

I think there is value in having more than one approach, and let the user choose which works for them. So, I would not remove DWB.

> If anyone has some insight on tangible pros/cons of each I'd love to have that documented for future readers and my education.

I haven't dug deep into it, and now that you ask this question, I want to look into it and find pros/cons.
I don't have much experience really testing them, besides using them in school projects, etc. 
If the porting is done before I get to researching the pros and cons, we can just test it and see it for ourselves! 

> I would also recommend if we're going to start promoting community projects to the nav stack, potentially we should split out repo for plugins separately so we don't go too wild here

I'm a little fuzzy on this. The TEB would live in the dortmund's repo as a plugin is what I thought.

mkhansen-intel, 2019-09-25:
Great to see you both jumping on this one. I agree that it should hopefully be a small effort, but I haven't looked at the TEB code that was ported. 

Just to clarify, the reason we are doing this is because we want a working example of an external plugin working with the nav2_core interface as a way of testing that our assumptions are valid. TEB makes the most sense because we have had others ask us if we're porting that one in the past.

I don't think we should pull it into our repo however, unless the previous maintainers are going to abandon it (I don't believe so).

SteveMacenski, 2019-10-12:
@vinnamkim's PR is good & will be merged soon. In terms of support we  can give, its been given and its just on the maintainers to polish and merge

closed 2019-10-12 by SteveMacenski

robotcopper, 2023-08-04:
Hello, i may have miss something but i am looking for TEB controller in the nav2 stack under humble since the [doc of nav2 ](https://navigation.ros.org/tuning/index.html#controller-plugin-selection) says that there is TEB. How ever i don't find the package. Are DWB and TEB  now mashed-up? 
I am looking for TEB to control a holonomic robot and to do obstacles avoiding.

SteveMacenski, 2023-08-04:
I'd recommend MPPI https://navigation.ros.org/configuration/packages/configuring-mppic.html the successor of TEB. thanks for the note, I'll update that page on Monday

robotcopper, 2023-08-05:
Thanks for this quick answer. I will try to find out how to work with it.