ros-navigation/navigation2 #933: nav2_dynamic_params no longer used: add to .deprecated directory?

SteveMacenski, 2019-07-16:
This code isn't being used as far as I am aware, can we move this to an auxiliary directory. I find for projects like this I like to add a `.deprecated` directory for old packages that chunks of code may be useful in the future or they may have a resurgence of usefulness.

crdelsey, 2019-07-17:
I'd rather just delete it. It would still be around in git history if we really want to bring it back. 

@bpwilcox @mjeronimo Was there a reason we kept the package around?

bpwilcox, 2019-07-17:
I believe we kept it around since we may refactor some of the code into nav2_utils when we bring back dynamic parameters into the lifecycle nodes. We've yet to investigate that though.

SteveMacenski, 2019-08-24:
Any update here?

ruffsl, 2019-09-17:
Resolved: https://github.com/ros-planning/navigation2/pull/1128

closed 2019-09-17 by ruffsl