ros-navigation/navigation2 #943: New pluginlib adaptors for Local/Global planning and costmap layers

SteveMacenski, 2019-07-17:
We should take queues from nav_core in nav1, as well as Locus's nav_core2, as well as what the move_base_flex folks have done

SteveMacenski, 2019-07-18:
I've just looked at nav_core2 and I think that's a really great place to start, what else would we want to add to it (if anything)? Maybe some primitives like spinning/backing up *hint hint, sounds like the recoveries can share that costmap/TF buffer inside the controller*

yathartha3, 2019-07-23:
@SteveMacenski I am looking at nav_core2. The first thing to do would be to define the interface we want, and then factor out common functionalities for local planners in nav2_util.

SteveMacenski, 2019-07-25:
I'm not sure exactly what you mean by common functionalities. nav_coreN is just an abstract class for pluginlib, its not really implementing anything. 

I would also recommend if there's no additional functionality we want, we may as well just use the nav_core2. To be pragmatic, I don't see Locus bending to our will, and in the absence of strong opinions on the matter (anyone?) may as well just join them to make porting to ROS2 easier for dlux & any additional controller/planners added.

SteveMacenski, 2019-07-29:
We also need to make a pluginlib wrapper for the recoveries, right now the recoveries have the base requiring filling out an action server, which is very restrictive and not necessary in line with some recoveries. The base recovery class isnt actually needed since there's no abstract factory, we just spin up the nodes in a main().

We should think about this plugin interface, move the recovery Action wrapper as a utility (or injust into the 2 action-based recoveries)

crdelsey, 2019-08-01:
@yathartha3 and I drew up some pictures of a possiblity for the local planner interface.

[image]

[image]

Basically, we'd take the main action server execute loop in dwb_controller and make it generic. It could load a plugin that implements something like @DLu's nav_core2 interface. We could make it so that the plugins could be lifecycle enabled or not.

In addtion, DWB has a goal checker plugin that is used to determine if we reached the goal or not. Instead of that being a plugin to the specific controller, we'd make that a plugin to the main action server loop as well, making them generic.

*Diagram Source*
[nav_local_planner_if_sequence.plantuml.gz](https://github.com/ros-planning/navigation2/files/3458152/nav_local_planner_if_sequence.plantuml.gz)

SteveMacenski, 2019-08-01:
100% agree, I had the same thought. Generic action interface that's exposing the action/services/topics and then a plugin loadable controller/planner which implements the algorithm and is given a TF buffer, costmap, namespace, etc. 

Though I'm not sure if we should make both a lifecycle and normal version, I feel like we should pick one and force people to stick with it. If they just have to implement `on_configure()`, `etc()` methods for the pluginlib definition and don't have to think about the logic behind lifecycle, that seems totally fair to me. Worst case the _dont_ and in `initialize()` they do everything, no harm no foul as far as navigation2 is concerned (other than a poorly implemented controller)

crdelsey, 2019-08-01:
> Though I'm not sure if we should make both a lifecycle and normal version

So you're saying "Local_Planner_Base" would have all 5 methods - initialize, on_configure, etc. The user can choose which they want to implement. They can do everything in initialize and just return in the on_* methods, or vice versa?

SteveMacenski, 2019-08-01:
Ah, looked at the lifecycle spec, I thought there was an `initialize` as well - nope haha.

I'm saying have the necessary state functions `on_*()` with an explanation in the nav2_core(2) plugin header what should and shouldnt be done in each. If someone chooses to ignore most of the states and just put everything in one of the functions, that's on them (and I would add a comment in that header about which function you should use for that case). 

Then you have the rest of the functional items to do the actual planning work. If some of the functions for planning can mix with the lifecycle functions, that's most ideal. Those lifecycle functions can be called anything we want, just under the hood they're implementing the lifecycle spec, there's no reason we need to really explain or expose to the user anything about lifecycle if we dont want to (and I dont think we should expose anything about it)