ros-navigation/navigation2 #949: Many signalFailure - Drop message warnings

crdelsey, 2019-07-18:
## Bug report

**Required Info:**

- Operating System:
  - 
- Installation type:
  - 
- Version or commit hash:
  - 
- DDS implementation:
  - 
- Client library (if applicable):
  - 

#### Steps to reproduce issue
Run nav2_system_test or just run the nav stack

#### Actual behavior

#### Additional information
There are a lot of messages from global costmap along the lines of
```
10: [world_model-5] [INFO] [global_costmap.global_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 2.400 for reason(0)
10: [dwb_controller-7] [INFO] [local_costmap.local_costmap_rclcpp_node]: [signalFailure] Drop message: frame 'base_scan' at time 14.400 for reason(0)
```

It seems these messages only come from the costmaps

mjeronimo, 2019-07-19:
In the obstacle layer there is a MessageFilter created when the plugin is initialized during onInitialize. All of the costmap_2d_ros plugins are initialized during costmap_2d_ros's on_configure method. When on_activate is invoked, the code is waiting for the transform to be available (base_link to map). However, the transform is not available yet because the user hasn't entered the initial_pose. So, the message filter's queue fills up and the error messages start when the message filter starts dropping messages. Once the user enters the initial pose, the on_activate method continues and the queue begins to drain. 

Part of the problem, IMO, is the "impedance mismatch" between the lifecycle-style interface (on_configure, on_activate, etc.) and the plugin interface (onIntialize, etc.). Another part of the problem is the user having to enter the initial pose during system startup.

crdelsey, 2019-07-19:
I"m seeing these even after setting the initial pose. Even while we are in the middle of navigating. I think they've gotten a lot worse with the latest ROS 2 code.

SteveMacenski, 2019-07-19:
To @mjeronimo's point: I dont think it's unreasonable on bringup to just publish a Identity transform for map->odom so that everything starts OK if not given a starting pose from the get-go. Since the filter should be always producing odom->base_link, that should give up closure on transforms in the active state even if the initial pose not yet set.

crdelsey, 2019-07-19:
>  I dont think it's unreasonable on bringup to just publish a Identity transform for map->odom so that everything starts OK

We specifically didn't do that because it messes up the costmap. As soon as that transform is available, costmap will start processing laser scans and putting obstacles on the map as if the robot were located at the origin. When the user then sets the initial pose, the robot's location changes, but all the obstacles stay in place.

This frequently results in an error right from the start, because the costmap is clogged up with all the false obstacles.

We need to fix costmap in some way to prevent this if we want to pretend we're at the origin on startup.

SteveMacenski, 2019-07-19:
Well 2 options I think of as knee-jerk reactions
- A) the "who cares" approach. If there's fake-obstacles in the map, as it tries to plan it'll fail and the recovery will clear the costmap and you're done
- B) the "hand engineered" approach, were on bringup, the very first initial pose you receive will clear out the costmaps if the map->odom transform is set to identity (and *not* do that if its been started with a real pose)

rotu, 2019-07-19:
Added a pull request in Geometry2 to make some of these error messages more descriptive https://github.com/ros2/geometry2/pull/130

crdelsey, 2019-10-01:
Problem is not causing actual problems right now and is related to the underlying libraries.

rotu, 2019-10-02:
@crdelsey I disagree. This *is* an actual problem. It's difficult to troubleshoot when the log is full of warnings you're supposed to ignore.

SteveMacenski, 2019-10-02:
I think what he's trying to say is that these aren't actual problems in this project, its from the libraries we use which have their own issue trackers

rotu, 2019-10-02:
@SteveMacenski @crdelsey if this is truly correct use of the library, please link this issue to the underlying bug in the geometry2 library.

crdelsey, 2019-10-02:
@rotu I'll reopen this. I think we can fix some of the error messages you are referring to. 

I closed this, because, when I filed the issue, I had a different problem in mind. We used to get lots of these error messages even once we were up and operating. Now, with the improvements in tf2 and FastRTPS, I seldom see this error after the system is up.

SteveMacenski, 2020-03-02:
@mjeronimo thanks for the analysis on root cause. It was pretty trivial to fix after your good analysis. See #1574

closed 2020-03-02 by SteveMacenski